ranenbg/Arduino-FFB-wheel #1: More detailed and visual diagram

lazenyuk-dmitry, 2022-04-06:
It's hard to understand the wiring diagram.
Could you make it more visual and detailed?

Like in this project for example https://github.com/vsulako/AFFBWheel

lazenyuk-dmitry, 2022-04-18:
Hey Milos. I want to draw a wiring diagram. What does "if LC - then clutch pot", "if LC - then handbreake pot", "if LC - then botton0" mean? what does LC mean?

lazenyuk-dmitry, 2022-04-18:
Why are you replacing pins with ProMicro and Leonsrdo? Leonsrdo also has pins 14, 15, 16 on the ICSP connector. Or am I misunderstanding something?

lazenyuk-dmitry, 2022-04-18:
What is i2C (ads1105) for?

ranenbg, 2022-04-19:
Hi Dmitry, tnx for the questions. Here is the explanation:

LC stands for load cell, as it is written in firmware_info.txt, all HEX that ends with number 2, has support for HX711 chip, in that case, analog input pin A3 is free and I am doing some rearranging of analog inputs due to that, such that I can use other available analog input pins for buttons for example.

I have to provide replacement pins for Arduino ProMicro, because it does not have pins 12 and 13 that I use in Leonardo. Even though Leonardo has this ICSP header with pin numbers 14-16, those pins are just shorted to corresponding SPI pins on the Leonardo which are 10,12 and some other - that is the part which you are overlooking.

My firmware supports external ADC or DAC with the i2C bus. There are two variants of external ADC available for Arduino. Those are 12bit ads1105 and 16bit ads1115. I'm using ads1105 to read pedal inputs and this is supported in HEX files that have letter "s" in the version. I did not compile such HEX yet, since no one was requesting it. Also I do now want to have to compile a milion versions of HEX files covering all possible combinations of my firmware options. I tried to compile only the most used ones. If one wants, I provided the entire source code and compile it himself or make any changes.

Tnx for all suggestions, I do try to provide all explanations and make them as clear as I can. All my instructions assume at least a basic knowledge of electronics and microcontrollers, since this is not a beginner project.

lazenyuk-dmitry, 2022-04-19:
> My firmware supports external ADC or DAC with the i2C bus. There are two variants of external ADC available for Arduino. Those are 12bit ads1105 and 16bit ads1115.

Are you sure you spell ads1105 and not ads1015 correctly?

I only found schemas for [ads1015 and ads1115](https://forum.fritzing.org/t/adc-ads1015-ads1115/4081).

lazenyuk-dmitry, 2022-04-19:
[image]

Hat switch this is a D-Pad?
I see where buttons 1, 2, 3 are connected, but I don't see where button 0 is connected.

ranenbg, 2022-04-19:
Yes, you are right it's ads1015. I made a typo, will fix it tomorrow.

Yes, the hat switch is an 8-way D-pad. Button0 is at pin A3 when LC is used (fw-v182) and on pin 4 when there is no LC, that is for all other firmware versions except for fw-v182. I will update the pinout png, tnx for pointing this out man.

lazenyuk-dmitry, 2022-04-20:
> Could you draw or write how you connected ads1105?

How do youuse ads1015?
More or less like this:
[image]

lazenyuk-dmitry, 2022-04-20:
Here's what I'm getting at the moment)
[image]

or* - connection differences for the firmware version.

But I think it's still complex, confusing, and hard to edit. I think we need to draw a diagram for each firmware version. Or draw one common to the major version and then draw only schema changes to other versions. What do you think about this?

ranenbg, 2022-04-20:
Hi Dmitry, tnx for this wiring diagram, it is correct. I must admit that I am not a fan of such wiring diagrams at all, that is why I tried to avoid making one and just provide the firmware pinouts. For me, an actual electronics scheme is enough, but for many people that is just way too technical, so I avoided that as well. I think the best would be just to keep the original pinout.png and add individual pictures for the wiring diagram for each external device separately, indicating which firmware supports it, just like you did for ads1015.

The ads1015 wiring diagram looks good, I'm preparing my own, which looks pretty similar to yours but with some useful details, I'll upload it soon.

As for the main/master branch stuff, I'm a total github noob as you can see. This was my first ever repo that I created, I was using gitbash and remote commands, so something got messed up and I never figured out how to fix that. I'd appreciate any help in solving it.

ranenbg, 2022-04-21:
Ok, I have uploaded the ads1015 wiring diagram and compiled a corresponding HEX. I also finally sorted out the branch things, such that there is only 1 branch called master.

lazenyuk-dmitry, 2022-04-21:
> Also, I do now want to have to compile a milion versions of HEX files covering all possible combinations of my firmware options, each time I update the code.

try to take a closer look at this project https://github.com/vsulako/AFFBWheel

It does not compile a million files, but simply writes what changes need to be made to config.h in order to get this or that functionality.

It might make your job easier. You can just flash from source through the IDE.

lazenyuk-dmitry, 2022-04-21:
Can you write to me what firmware versions you want to compile now and what they are for?

For example: fw-v18Xh - hat switch, fw-180 - main firmware and etc.

I want to understand whether it is possible to draw one general diagram and then draw diagrams that reflect which modifications. I think it should work)

lazenyuk-dmitry, 2022-04-21:
And I still really like this implementation of the buttons. There are a lot of them, 32 buttons. They are divided into two lines. For example, it is very easy to place 8 buttons on the steering wheel and 8 on the base, and also make a gearbox out of 8 buttons and place 8 more buttons on the gearbox. That's more than enough buttons and doesn't need to run a lot of wires. And it's cheap on the 74HC165. Do you plan to implement something similar on cheap components 74HC165 or analog?

[image]

lazenyuk-dmitry, 2022-04-22:
z-index supported in fw version with "z" suffix only?

lazenyuk-dmitry, 2022-04-22:
A4, A5 - button1,2 its only for fw-v180 or and forfw-v18X?

ranenbg, 2022-04-23:
Hi Dmitry, I see you have many question already for me, I'll try to answer.

I want to provide HEX for my firmware in order to minimize the effort required by users with no Arduino IDE knowledge. My firmware is using 4-5 Arduino usb core files which are modified and in order to compile from IDE is required to replace those files, which is not a trivial task at all. I have provided all modified files, used libraries and a full package or Arduino core files for boards version 1.6.21 so advanced Arduino users will be able to use that. For everyone else HEX is easy to upload with XLoader.

All info about what each firmware version support is in the ver_info.txt and ver_info_promicro.txt files. It is possible to make a wiring scheme for the most basic firmware version fw-v180 and make additional ones with specific devices separately.

As I mentioned before and I have written in my instructions, the code for button box that is uploaded to Arduino Nano is simulating shift register chip, so you can in principle just connect two 74HC165 in series and get 16 buttons readout, on firmware fw-v181. With a simple number change in my source code, more shift register chips can be supported, for now HID supports up to 28 buttons (4 of those are taken by HatSwitch, but that can be extended if required).

Buttons 1,2 on pins A4,A5 on Arduino Leonardo are enabled for all firmware versions. Only for Arduino ProMicro, since it does not have those pins, I have moved them to other available pins.

FFB clipping LED is to be connected at pin13 (Leonardo) and pin3 (ProMicro). This pin provides a separate PWM signal (it is not the same as for FFB output) to drive the LED and is not just a simple ON/OFF indicator. I have made a special function for it and this is how it works -  when FFB magnitude is from (0-89)% LED is off, for FFB (90-99)% it lights up linearly from (0-50)% of LED intensity and for 100% FFB it's at the max intensity or LED pin is at 5V. You have to connect LED anode (+) to pin13 and LED cathode (-) to one end of a series resistor for example (220-680)Ohm and another end of a resistor to GND. I have written how FFB clipping LED works in one of the firmware info txt files.

lazenyuk-dmitry, 2022-04-23:
> As I said earlier, I'm not a fan of such wiring diagrams because they have no value to me, I do not want to have to trace some lines and colors to be able to know what goes where. I do think that a simple pinout table for the firmware is good enough and provides complete wiring information.

Thank you for your responses. I agree with you that drawing a wiring diagram is not very convenient. And perhaps even for those who are well versed in electronics, the pinout table is much more convenient than the wiring diagram. But as a person who knows how to simply solder two wires, I’ll say that the wiring diagram is simpler and clearer for me, and the pinout table raises many questions and takes a very long time to figure it out, I even thought to draw a diagram just for myself.

lazenyuk-dmitry, 2022-04-23:
I have almost everything ready. I will prepare all the changes and make a pull request.

lazenyuk-dmitry, 2022-04-23:
I have a question about version 183, you write "but has analog FFB output with 2x 12bit DAC MCP4275". I understand that this is the only difference. Can you provide a pinout table for the DAC MCP4275 or a wiring diagram?

[image]

ranenbg, 2022-04-23:
Yes, the connection is something like that, except that SDA is at pin 2 and SCL at pin 3. I'll make a wiring diagram for this one and put a differential amplifire to utilze 2xMCP4725 in DAC+- mode, such that output is +-10V which is suitable for AASD.

lazenyuk-dmitry, 2022-04-23:
> This was my first ever repo that I created, I was using gitbash and remote commands, so something got messed up and I never figured out how to fix that.

You can try using a GUI like *[SmartGit](https://www.syntevo.com/smartgit/)*.
Of course, it's good to be able to use git Bash, but the GUI is convenient)

lazenyuk-dmitry, 2022-04-23:
> I'll make a wiring diagram for this one and put a differential amplifire to utilze 2xMCP4725 in DAC+- mode, such that output is +-10V which is suitable for AASD.

 Ok. I just have to make a circuit for the MCP4725.

lazenyuk-dmitry, 2022-04-23:
[image]

Is it correct that I connected the PWM direction(pin11) to R_EN and L_EN pins on BTS7860?

ranenbg, 2022-04-23:
Tnx for suggestion about SmartGit, I'll check it out.

No, you should not connect bts pwm enable pins to pin11, because in pwm+- mode, this pin is allways low (unused), while in pwm+dir mode it is low when ffb is negative and high when ffb is zero or positive. It should be connected to 5V, the same ss vcc pin of bts.

lazenyuk-dmitry, 2022-04-23:
> No, you should not connect bts pwm enable pins to pin11, because in pwm+- mode, this pin is allways low (unused), while in pwm+dir mode it is low when ffb is negative and high when ffb is zero or positive. It should be connected to 5V, the same ss vcc pin of bts.

[image]

That's right? VCC connects to R_EN and L_EN on BTS7960?

Then what is pin 11 on the Arduino Leonardo for?

ranenbg, 2022-04-23:
Yes, this bts connection is good now.

Some motor drivers like brushless, H-bridge or AASD for servo motors have the input defined as pwm signal and direction signal. For that use case I made pwm+dir mode of FFB output, where  pin 9 provides pwm signal and pin 11 provides a direction signal.

ranenbg, 2022-04-28:
Hi Dmitry, I have uploaded a new wiring diagram for mcp4725. If all is fine, you can close this thread.

lazenyuk-dmitry, 2022-05-03:
I made a pull request)
https://github.com/ranenbg/Arduino-FFB-wheel/pull/5

closed 2022-05-03 by lazenyuk-dmitry