ranenbg/Arduino-FFB-wheel #4: Have a ads1115 16 bit and arduino uno button box

Armagedon13, 2022-04-30:
Hi and thanks for this firmware, my question is if i can have a ads1115 16 bit [ads datasheet](https://www.ti.com/lit/gpn/ads1114) and a arduino or multiplexer button box and if i can make a button matrix to optimize the buttons in pins like this:
[image]
And the multiplexer is the 74hc4067 or 74HC595 [74hc4067 datasheet](https://www.ti.com/lit/ds/symlink/cd74hc4067.pdf) / [74HC595 datasheet](https://www.diodes.com/assets/Datasheets/74HC595.pdf) and you can wire multiple by cascading:
[image]

Armagedon13, 2022-04-30:
Something off topic you can add more flash memory with this: [https://github.com/Marzogh/SPIMemory](https://github.com/Marzogh/SPIMemory) and a cjmcu-2516 module (W25Q16BVSIG chip) add 16 M-bit more memory

ranenbg, 2022-05-05:
Hi, I can compioe a HEX firmware with some custom options, that would be fw-v181hs. It will have support for ads1015 and arduino nano button box for 16 buttons. I did not enable ads1115 because it is much slower than ads1015, but I can if you wish, however you should note that only brake axis is 16bit in HID while other pedal axis are oboy 12bit.

closed 2022-05-07 by Armagedon13

ranenbg, 2022-05-11:
In order to compile my code yourself, this is a tricky part. It took me about 6 months just to compile Fernando's code back when I was starting. You can use the latest Arduino IDE 1.8.19, but I use older boards version 1.6.21. You will have to copy all the avr/core files that I provided and also replace a few USB core files, which are inside br_wheel_my folder. Said files have to be replaced in C:\Users\milos\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino. You also have to put all required libraries that I provided in your documents\arduino\libraries folder.