ranenbg/Arduino-FFB-wheel #7: Esp32 Upgrade

etempm, 2022-06-20:
Hi, thanks for great work. I want to convert project to esp32.   I can code esp and arduino and want to convert project to arduinoesp framework. What is your opinion about this ? Can be doable, easy or hard ? Thanks again.

ranenbg, 2022-06-20:
Hi man, you are welcome. I think that porting this code to esp32 is an extremely difficult task, but feel free to do it :) There are many Arduino USB core files that I modified. Esp32 will have those very different, so you will most probably have to write them from scratch. Remember, this is low-level C++ stuff, so you have to know your way around HID very well.

etempm, 2022-06-20:
Thanks for pointing me to right direction. I am no so pro low level c++, so I will go with arduino. I have a small problem when compiling. Arduino ide can not find Platform.h file. Where can I find it ? I am using Lenoard.

Thanks for quick reply.

etempm, 2022-06-20:
Finally One more question. VNH5019 is very pricy in my country. I can easyly find BTS7960 here and it is cheap. Can I use this firmware with BTS7960 ? Thanks again.

ranenbg, 2022-06-20:
Yes, as you can see in the description I have provided wiring for BTS7960. What you saw in the code is just some leftover from previous guys that worked on the code, I left it there, but it is commented and VNH5019 is not in use in the code.

ranenbg, 2022-06-20:
Here is some info on what you need to do, that I know so far: