ranenbg/Arduino-FFB-wheel #11: Encoder Resolution

etempm, 2022-07-31:
Hi, what is your suggested resolution in ppr? I have 400 and 600 ppr with z index. Also I have as5047p magnetic higher resolution. Is bigger is better, or there is a limit for arduino?

I am making some modifications to work with dd wheel foc controlled. If I succeed, I can share it.

Thanks again.

ranenbg, 2022-07-31:
Hi dude, in principle the bigger the encoder PPR the better, but there is a limit in arduino, regarding how fast it can detect pulses at interrupt pins. This limit is at about 100kHz and very much depends in the max desired encoder rpm. Magnetic hall can be good if 14bit, 12bit is too low, but you have to implement it in the code yourself. There is not enough mcu power to run the FOC loop along side everything I have so far. Even stm32f407 would struggle.

etempm, 2022-07-31:
Foc is working on stm32, simplefoc. It takes inputs from arduino(your ffv).
For optical omron like encoders, so 400 - 600 ppr is not good enought, right?

ranenbg, 2022-07-31:
It is good enough. If you don't mind losing the z channel, you can allways mount encoder via a pulley to motor shaft and increase CPR, you don't need more than 5k, 6k PPR.

closed 2022-08-01 by ranenbg