ranenbg/Arduino-FFB-wheel #14: Force feedback not working

tomascarabajal7, 2023-05-26:
Hi! Great project!
I'm currently having issues with 2 subjects.
1) my force feedback doesn't work. Hardware wise it all moves. But when i try the wheel config app or the i racing one nothing moves or changes. I have everything conected as it is specified. The only time i got it to respond was when i plugged one of the pwm to pin 11 and it made sort of a barrier just in one way. This brings me to my second question.
2) PWM pin direction (or pin 11) is not specified ANYWHERE. I see there is an option in the software tl implement it but hardware wise I don't even know how to connect it. 
Thanks in advance!

ranenbg, 2023-05-26:
Hi man, tnx. So to continue answering from my other GUI repository where you asked questions firmware related.

1. You have to upload the correct firmware (HEX file) to your Arduino first. Please have a look at the readme of the FFB-wheel repository, there you have a detailed explanation of how I name each HEX and what it means, which features it supports. For you, I assume something like HEX with fw-v220 would be a good start. As I wrote in the readme, please make sure to upload the correct HEX that is designed for the Arduino board that you use. Micro HEX is only for Arduino ProMicro board, while Leonardo HEX can be used for both Arduino Leonardo and Arduino Micro.

2. Once you upload for example fw-v220, you can configure it using my GUI the desired ffb output option, for that I also made a dropdown menu selection in the GUI, just note that for each change of PWM settings you have to restart Arduino for changes to take place. Depending on which motor type and motor driver you plan to use. I assume you have a DC motor and use it with H-bridge driver like BTS7960, so then just have a look at readme, there I have put a detailed wiring diagram for this case. In the very first image in the readme, you can see the pinouts for the firmware, by default PWM+- mode, you will have pin9 as left PWM signal and pin10 as right PWM signal. If you want to utilize the direction pin then you have to select PWM+dir mode in my GUI, where PWM signal is on pin9 and direction signal will be on pin11. Note that I made firmware in a way that configuring everything in Arduino takes place via serial communication, so in principle you could use any other software cable of serial configuration to configure it, you don't have to use my GUI if you don't want. Then I would refer you to RS232commands.txt for more info about it.

I hope this answered all your questions, but if you have more, I'll gladly answer. Take some time and read the manuals, txt files and readme more carefully, I have explained pretty much everything there.

tomascarabajal7, 2023-05-26:
Excellent answer! Thank you very much :) 
I understand now that i hD the wrong firmware. Apparently i was so tired that night that choose the wrong one hahaha. 
Besides that i still don't understand how the direction pin works at a hardware level 🤔. I have a bts 7960 and I've wired it exactly as the diagram shows.
Also I'm new to ffb wheels so there are some things that i don't quite understand. For example, if i choose "c" (hardware wheel re center support, do i need to wire some sort of switch? Or the program will remember where i set las time the center and use that?.
Sorry if this seems silly but my wheel almost killed me last night hahahaha

ranenbg, 2023-05-27:
The "c" version firmware gives you two ways to reset your wheel's center position (0 angle), by hardware using a switch, or by software in GUI by pressing center button. As this firmware doesn't utilze a 3rd channel of an encoder called z-index you will have to center the wheel after each powerup. If you do have an encoder with 3 channels or with a,b and z you could use "z" firmware version, there it would automatically find 0 angle as soon as a z-index pulse from encoder is detected.

For your bts7960, I recommend the following PWM settings: fast top, 8kHz and PWM+-. You should not use PWM+dir mode with it, as this mode is designed for other type of motor drivers like ac servo motor driver or some other that need that kind of control input. Briefly, PWM+dir mode works in a way that PWM provides and absolute magnitude of the FFB signal (always positive) and dir signal gives the sign (direction) of the FFB signal. In my firmware, on dir pin 0V - left direction and 5V - right direction, while PWM's duty cycle represents 0-100% of FFB signal.

Just take your time and be patient, once you setup your wheel correctly you will enjoy it very much. It's very rewarding feel and this effort pays off in the end.

tomascarabajal7, 2023-05-30:
Thanks again for the great answer 👏😁. 
I made it work and it's awesome! In some games its dull but older ones like assetto corsa or my fav Race Room it's EPIC how i can feel every detail! 😍🥳🥳🥳. 
I still need to figure out a way to keep the belt tensioned or it skips steps 😅. I think i will try printed gears instead. Any advice? 
Also the version with the nano wheel works FANTASTIC!. 
I'm from Argentina and wheels with FFB are soooo expensive so I'm enjoying this thing A LOT!

ranenbg, 2023-05-30:
You're welcome. I strongly recommend you to try LFS (live for speed), in my opinion this is the best sim with the best ffb and demo version is free, there are a lot of servers to chose from, plenty of players from south america. Assetto corsa does have a decent ffb, but LFS has even better while online gameplay and overall game experience is many levels above.

I tried using gears. The only gear type that makes sense for this application is a Herringbone (double helical) gear, but those are very difficuilt to 3D print correctly. It may be better to stay with belt + pulley system, because it allows you to have some imperfections in the mechanism. People make spring loaded dummy pulley made out of M5 or M5 screw and some bearings to keep the belt tightened. Try HTD 3M or even 5M with 3mm or 5mm tooth spacing, these are timing belts and pulleys with l, that DIY community uses the most. Usually people buy small 12 tooth aluminium pulley that goes on motor shaft and design and 3D print their own big pulley with 60+ teeth.

By wheel with nano, I think you are talking about a button box firmware for arduino nano which allows connection of up to 16 buttons. Yeah, it took some effort to make that working correctly and not waste too much of the MCU time in the main board firmware.

tomascarabajal7, 2023-05-31:
Thanks for all the info! I will check that sim out! 
Im using a 12tooth and a 60tooth. I'm thinking of printing a few gears for the encoder. I was able to tighten the belt for the motor but it was a pain in the ass haha 😆. 
Now i get all range, no skipping and even the psu has a hard time keeping up (12v 10A) with the dc775. 
I think i will add a second motor but donno what to do with the psu... Maybe a pc one? 😅. 
I will still try gears at some point cause in my country pulleys are hard to come by, expensive and more so for thicker ones. 
And so if it is meant for Thrustmaster... Will it be possible to use Thrustmaster wheels? Just for curiosity haha.

ranenbg, 2023-05-31:
Please, let's keep the disscussion only to firmware related things. 

In principle Thrustmaster wheel rims should be compatible with my firmware, but one has to be very careful because TM wheel rim operates at 3.3V and Arduino is 5V, so some bidirectional TTL level converters must be used for every signal line between arduino and a wheel rim.

closed 2023-05-31 by ranenbg