ranenbg/Arduino-FFB-wheel #15: Wheel Control 2.1 doesn`t show FW version.

RomanFolab, 2023-06-09:
Wheel Control 2.1 doesn't show the FW version. 
Was uploaded "brWheel_my.ino.leonardo_v14"
Im using Arduino pro Micro

ranenbg, 2023-06-09:
Hi, if you look closely into the readme of GUI in my other repository, you will see that wheel control v2.5.1 is only backward compatible with Arduino HEX firmware down to v170. The HEX version you have now v14 is ancient, this was way before I introduced the firmware recognition features. You have to update to the latest firmware, for example, v220m, this is suitable for your Arduino ProMicro. Please have a look at the pinout diagram to avoid wiring confusion.

RomanFolab, 2023-06-09:
Hi 
I have uploaded v220m and received from wheel control next massage 
[image]

closed 2023-07-05 by ranenbg