ranenbg/Arduino-FFB-wheel #17: pwm frequency question

ernest-cpu, 2023-07-06:
What is the max pwm frequency and resolution in your firmware? I see in the data sheet the best I can get out of the 32u4 is 9.6bit and 20khz pwm frequency on a 16mhz pro micro. It makes sense to use 20khz because you don't want a really whinny motor. Does it make sense?

ranenbg, 2023-07-07:
Hi dude, yes it does make sense. I have put detailed information about the available PWM frequencies of my firmware in the documentation. Have a look at firmware_info.txt in the docs folder, there are plenty of PWM options to choose from.

ernest-cpu, 2023-07-07:
How does fast compare to phase correct? I would imagine that it might cause the wheel to be more noisy but the forces should feel finer?

ernest-cpu, 2023-07-07:
I was imagining 800 steps in one direction not dived in half haha

ranenbg, 2023-07-07:
Fast PWM has random small glitches even when there is no PWM signal. This is however no issue at all for DC motors with bts7960, thay anyway require some min voltage to start spinning.

It is 800 steps per direction, since there are two PWM channels which are used in PWM+- mode.

korvolle, 2023-07-08:
Im

> Fast PWM has random small glitches even when there is no PWM signal. This is however no issue at all for DC motors with bts7960, thay anyway require some min voltage to start spinning.
> 
> It is 800 steps per direction, since there are two PWM channels which are used in PWM+- mode. One channel is for one force direction.
> 
> When using PWM0.50.100 mode, there the PWM steps are shared for both directions, as 50% duty cycle is for 0 force, leaving only half of steps for each direction, but the advantage is that one only needs this 1 PWM channel (no dir signal is required).

I have a motor controller that only accepts "RC PWM". Would this work to connect to that controller? Sorry for this dumb question, im not good at electronics at all

ranenbg, 2023-07-08:
It's not a dumb question at all. Sometime in the past I have added this special PWM mode for RC servos or RC ESC, I call it in my firmware RCM.

You can use it for your motor driver, there are plenty of frequencies to chose from. Normaly, in RC world they use 50Hz for this PWM signal, which is good enough for human control (20ms update rate), but it's very low for FFB signal. In theory, max supported frequency for RCM mode is 500Hz which is exact match with the FFB update rate of my firmware. Since I can't check all possible drivers myself, you may want to experiment with your driver and see which frequency it likes the best. I wouldn't go lower than 100Hz.

korvolle, 2023-07-08:
Thank you so much for the reply! Its actually a hoverboard motor driver that i will use for this project. I will update with how it goes once i get it working :)

korvolle, 2023-07-08:
Btw i just noticed that your firmware has support for DAC output and the custom firmware that i will use on the hoverboard controller has support for ADC input, wouldnt that work better than PWM you think?

ranenbg, 2023-07-08:
Ah I see, good. I think that you will need to calibrate the driver, such that it knows which PWM duty cycle corresponda to 0 force and which for max left and right forces. You can use autocenter spring for this, with enabled ffb realtime monitor amd just move encoder untill you get desired ffb outputs, for the driver calibration procedure. As I'm told, this is quite tricky to do. So pls, do update

ranenbg, 2023-07-08:
Yes, you can use external dac. I don't know which is better, you may want to play with this as well. PWM signals are less prone to external noise. If you go the analog way, you will have to pay special atention to shielding all the wires that cary ffb signals, in a similair way as you would do with audio signals before the amplifire.

closed 2023-07-24 by ranenbg