ranenbg/Arduino-FFB-wheel #18: Dual bts7960

tomascarabajal7, 2023-07-07:
Hi i have an issue with my dual motor Setup. 
I've connected 2 bts7960 to two dc775 and every wire that comes from the Arduino in parallel. And when it hits an end (either right or left) it's starts to spin like crazy randomly. But if i use spring or centering effects it just works. The problem it's, for now, only on the ends. 
I believe there could be 2 reasons but please enlighten me 😅. First one could be that there is some sort of lag between the two drivers. 
And the second one could be the wiring. In this one i will like to add that for the first driver i used the shunt resistors you specify on the wiring diagram and in the second one i didn't bother. Could this be the issue? What are this shunt resistors used for? 
Thanks in advance 👏🤩

ranenbg, 2023-07-07:
Hi man, yeah, interesting behaviour. I recommend to first use only 1 bts7960 and 1 RS775 motor, untill you make it work as intended. Then you may try adding one more bts in parralel to PSU and to Arduino PWM output pins. Endstop force is very similair to a spring force, the only difference is that it gets activated when the encoder hits the ends of the axis, the force should

tomascarabajal7, 2023-07-07:
Thanks! So quick! Haha. 
I had the wheel working with two motors and one bts but the bts burned one of the chips so it only worked on one side haha. 
I will check the cables again but everything is jumped from the connection i had that worked fine. Any othe ideas?
And i will take the shunt resistors off then 😅. I misunderstood from the diagram that i SHOULD have them connected.

tomascarabajal7, 2023-07-07:
I'm using exactly that, 12v 40A and keeping it to 0.7 overall strength. And from what I've researched 775 has a max of 15A, so two of them would be safe for a 7960.. but i was wrong apparently haha. Also both are cooled with a blower so as they don't spin they are still cooled. 
I donno really where this issue could be coming from... Again, I'll try cables.. but could it be that the two bts have a delay? I'm loosing my mind haha. If i plug just one it works fine.

ranenbg, 2023-07-07:
I'm not exactly sure, but I think that stall curent for RS775 is much higher than 15A. It is a 300W motor, so it depends on the voltage. It's coils should be good even at 30A. The issue is that in FFB wheel application motor is barely spinning and is not able to cool its coils with airflow created by the fan at the back of the shaft. 

Also the issue with bts is that it comes with a horrible cooling solution. You may think that it's fine as heatsinc is not even warm, but the fact is that heat is not conducted well from mosfets, causing them to overheat and die long before heatsinc gets warm.

tomascarabajal7, 2023-07-07:
Hahahaha yep! I have 80mil fan just for the driver and also did glue some of those small heatsinks that comes with 3d peinters but with 2 motors it just died wile racing hahaha. 
Now the solution and also the problem is to use 2 of them, one for each motor... But after taking out resistors and checking wires it goes crazy when it touches the end 😅💔

tomascarabajal7, 2023-07-07:
Solved it! 👏😍. 
Problem was... Wait for it... Damper Gain in wheel control 😅. It was in 0.6, lowered it to 0.3 and done! I also changed the Min torque % to 5.5 As you recommended. That for sure did something but to be honest i don't really know what most of this values do ✌️🤣. If you can explain this one it would be great. 

Again... Im so happy with this project and your kindness to always help so quickly. You have earned heaven my friend! Have a nice weekend!

ranenbg, 2023-07-08:
Tnx man. Unfortunately, I don't have much time for my hobies at the moment, so I will not be writing a detailed explanation for every slider and button in my gui. I tried to make it in a way to be very intuitive to use and self explanatory.

The beat is to first read manual.txt for the gui. There I explained all important stuff and how to use it. In principle, every slider on the right is a gain for each FFB effect and there is a possibility to enable some of those effects as "allways on", also called user or desktop effects. In my gui, you can enable/disable these with a square button next to a slider. These are damper, inertia, friction and auto center spring. For a DC motor, I do not recommend to use any of those. You can leave auto center spring enabled, because thos one is special in a way that it is automaticaly turned off, as soon as a game or other application takes over the ffb control.

tomascarabajal7, 2023-07-17:
Hey! I've been doing some testing... Apparently the problem is that the motors spin too fast. This i should fix it by changing the gear ratio from 4:1 to 8:1 or something like that. 
The motors without the actual wheel attached to the shaft spun out of control. Once there is some weight to it doesn't go crazy. 
Besides that i believe there is something to add to the firmware. The force feedback should cut after a certain threshold. Otherwise if the system looses position it just keeps spinning forever and braking everything into pieces. I believe, if possible, this could add some layer of security. 
Maybe something simple as a condition that contemplates some extra margin after the set maximum (for example if you set 900deg it should be 50 deg extra, or something like that) and when exceeded the ffb dies. 
Something to add more security. 
Thank for your time! I hope this helps :)

ranenbg, 2023-07-17:
The question is where you mounted encoder. Arduino has a certain limitation above which it can no longer detect pulses from encoder. But yes, you are right there could he some sort of software detection of curent angle and a safety that cuts off ffb in case of emergency. Usualy these stuff are handeled by the motor driver, but since bts is simply a brainless amp, it must be done somehow with arduino. I'll think about it and see what can be imolemented.

closed 2023-09-13 by ranenbg