ranenbg/Arduino-FFB-wheel #19: doubt

Zigandbee, 2023-07-24:
I'm trying to use open programming but it's giving me this error in Arduino IDE

Arduino: 1.8.19 (Windows 10), Placa:"Arduino Leonardo"

In file included from C:\Users\valdemir\Desktop\arduino-1.8.19\Arduino-FFB-wheel-master\brWheel_my\brWheel_my.ino:27:0:

ffb.h:297:3: error: 's32' does not name a type

   s32 rampStart, rampEnd; //milos, added

   ^

ffb.h:300:3: error: 's16' does not name a type

   s16 magnitude, positiveSaturation;  //milos, added positiveSaturation

   ^

ffb.h:301:3: error: 's16' does not name a type

   s16 offset;

   ^

In file included from C:\Users\valdemir\Desktop\arduino-1.8.19\Arduino-FFB-wheel-master\brWheel_my\brWheel_my.ino:28:0:

ffb_pro.h:8:7: error: 's32' does not name a type

 const s32 SPD_THRESHOLD = 0; //8

       ^

ffb_pro.h:9:7: error: 's32' does not name a type

 const s32 ACL_THRESHOLD = 0; //milos, added

       ^

ffb_pro.h:10:7: error: 's32' does not name a type

 const s32 FRC_THRESHOLD = 1; //milos, added - friction treshold

       ^

In file included from C:\Users\valdemir\Desktop\arduino-1.8.19\Arduino-FFB-wheel-master\brWheel_my\brWheel_my.ino:28:0:

ffb_pro.h:59:5: error: 'f32' does not name a type

     f32 Update(s32 new_pos);

     ^

ffb_pro.h:61:5: error: 's32' does not name a type

     s32 mLastPos;

     ^

ffb_pro.h:62:5: error: 'f32' does not name a type

     f32 mLastSpeed; //s32

     ^

ffb_pro.h:63:5: error: 'b8' does not name a type

     b8 mLastValueValid;

     ^

[... 1520 lines cut ...]
   ^

pwm:21:3: error: 'RCM_max' was not declared in this scope

   RCM_max *= RCMscaler(pwmstate); // milos

   ^

C:\Users\valdemir\Desktop\arduino-1.8.19\Arduino-FFB-wheel-master\brWheel_my\pwm.ino: At global scope:

pwm:57:25: error: variable or field 'activateFFBclipLED' declared void

 void activateFFBclipLED(s32 t) {  //milos, added - turn on FFB clip LED if max FFB signal reached (shows 90-99% of FFB signal as linear increase from 0 to 1/4 of full brightness)

                         ^

pwm:57:25: error: 's32' was not declared in this scope

exit status 1

's32' does not name a type

Este relatório teria mais informações com
"Mostrar a saida detalhada durante a compilação"
opção pode ser ativada em "Arquivo -> Preferências"

ranenbg, 2023-07-24:
Hi dude, first of all, it is pointless to post such a long compiler error list. I see that you didn't read my reply for your question about how to compile the source code. I'm copy pasting the instructions I wrpte in the readme file of the repository.

How to compile the source
In order to compile the firmware yourself you may use windows7, 8, 10 or 11, you need to install Arduino IDE v1.8.19 and Arduino Boards v1.6.21. You must place all [libraries](https://github.com/ranenbg/Arduino-FFB-wheel/tree/master/arduino-1.8.5/libraries) in your .../documents/Arduino/Libraries folder. In windows folder options set to show hidden files and folders then navigate to C:\Users\yourusername\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores. Rename the folder "arduino" as a backup as we will need some files from it latter, I just add "arduino_org" to the filename. Create a new folder called "arduino" and place the entire content from [modified core](https://github.com/ranenbg/Arduino-FFB-wheel/tree/master/arduino-1.8.5/hardware/arduino/cores/arduino) into newly created "arduino" folder. Navigate back to "arduino_org" folder and copy files "IPAddress.cpp", "IPAddress.h", "new.cpp" and "new.h", then paste and replace the ones inside the "arduino" folder. That should fix all errors and you should be able to compile the code. Bare in mind that if you make any changes to HID or USB core files you will need to repeat the procedure and paste all modified files in the newly created "arduino" folder each time.

ranenbg, 2023-07-26:
Make sure to install Arduino boards 1.6.21 and copy and paste all the modified files to the correct location. Also make sure that in your Arduino IDE sketch, you have all the tabs with .h and .cpp files.

closed 2023-08-24 by vortex464