ranenbg/Arduino-FFB-wheel #22: Reduce torque

jwinfieldnz, 2023-07-29:
Great project! I built a steering wheel using a cordless drill. It seems to have worked out of the box which is great. This is my first project with FFB and am a bit lost so apologies for the N00B questions. Being a cordless drill it has finger amputating torque. Which would be the best setting to tone down the power?

[image]

ranenbg, 2023-07-29:
Hi man, very nice build, tnx for sharing. Master control for overall ffb strenght is general gain slider, about 0.5 should be good starting point.Play with it first to find a good feel, then you can add a little bit of min torque slider to compensate for the ffb deadzone in low forces, about 4-5% should be enough.

jwinfieldnz, 2023-08-02:
Great I found it. Here is a video of my project. Not an issue technically: https://youtu.be/vo7tVuBD-hI

closed 2023-08-02 by jwinfieldnz

jwinfieldnz, 2023-08-03:
[Hackaday](https://hackaday.com/2023/08/03/force-feedback-steering-wheel-made-from-power-drill/) picked it up as well so that is cool.