ranenbg/Arduino-FFB-wheel #23: open

vortex464, 2023-08-02:
hello, my project needed a special firmware so I modified yours but I have a question for you to send the configuration data did you use 100% serial connection? How did you make the Arduino not crash with too much information. I also modified the force feedback to peg to get the "torque" I used the already existing torque variable is it the best option?

if (torque >= 0) {
        motorffb.TurnRight(abs(torque));
      } else {
        motorffb.TurnLeft(abs(torque));
      }

ranenbg, 2023-08-02:
Hi, congratulations on compiling my firmware, you are the 1st one  so far who managed to do it :)

I'm not sure that I understand what do you want to do with your project. Yes, I'm using virtual serial port connection to configure everything and send the ffb single value data, for the purpose of the feature of my gui called ffb real time monitor. Arduino is receiving serial comands that are terminated by CR, processing them and then sending the response terminated by CR+LF, there is no reson to have a crash. Each config data is sent individually. The difference is only in ffb monitor or torque data, once you send the comand to arduino it starts sending data continously, which requires some smart coding on the receiver side, you can have a look how I solved that in my gui.

vortex464, 2023-08-02:
It was exactly what I needed to modify the FFB, I just wanted to know if I collected the correct value I used the value I found "torque",. in serial communication I already understood!

closed 2023-08-24 by vortex464