ranenbg/Arduino-FFB-wheel #24: Hall effect support

jwinfieldnz, 2023-08-03:
I am not sure if it is tuneable in Arduino-FFB-gui but I would like to use hall effect sensors for the pedals. The problem with these (the ones I have anyway) is the quiescent voltage is around 3.2v and goes up maybe 1-1.5v to 4.2-4.8v. I was going to use a rail-to-rail op-amp or even a slave Arduino to convert the voltage but if this could be done in Arduino-FFB-wheel that would be awesome!

ranenbg, 2023-08-03:
Hi, you can use firmware with manual pedal calibration, there you can set the range for each pedal axis using the gui. Arduino analog input will read 0V to 5V in 10bits, so you should be use those hall sensors. The only downside would be that once you do the range calibration you can be left with a small number of steps, but even with 200-300 steps left out of 1024 it shpuld be still usable. Logitech pedals are 8 bits anyway (256 steps).

jwinfieldnz, 2023-08-03:
Actually, I just clicked on autocal and worked perfectly! 4096 steps. Hate to go OT but one niggling issue is the wheel in the GUI tool turns about 25% faster turning one way than the other. Maybe it is the encoder I am using.

ranenbg, 2023-08-03:
Ok, good, pedal autocalibration is done on every powerup, it just dynamicaly looks for min and max values and keeps the calibration only while arduino is powered. On each powerup, you just press all pedals full range and it will calibrate itself.

Yeah, that is interesting situation. You should check in DXtweak and maybe have to do reset device default calibration, encoder axis is Xaxis and it could be that something went wrong there. Then in gui you only have to set correct encoder CPR, desired rotation degrees and recenter.

jwinfieldnz, 2023-08-22:
Seems unless I have the encoder sensor perfectly parallel to the encoder disk it will turn further one way than the other. It must have some sort of parallax error.

closed 2023-08-22 by jwinfieldnz