ranenbg/Arduino-FFB-wheel #27: SPI version of button box

jwinfieldnz, 2023-08-26:
It would be great to have a SPI version of the button box firmware. So rather than use "DT," "CLK," and "PL" you could use SPI and then use something like a NRF24L01 to communicate between the wheel and the Leonardo removing the requirement to try and run wires up the steering shaft.

ranenbg, 2023-08-26:
Hi, I agree, it would be great for sure :) I'm not going to be  implementing this, I don't think here is enough memory left. You may go ahead and add it ofcourse.

ranenbg, 2023-08-27:
I did have a look into this, to maybe try to get the NRF module working on the Arduino nano, but again, even nano has SPI pins used for the shift register communication, so I really see no easy possible way to add this module anyway.

RaceMax, 2023-08-29:
Hi there. What about if, connect 2 slave boards to the master board? Or even use separate boards for different tasks with additional power supply ( plug in to one usb hub to connect PC) ?

Warning - I'm a noob yet) But  it's intereting project, and I get started ).. , now stuck with choosen fw version for fist prototype...

ranenbg, 2023-08-29:
Haha, you wild ideas RaceMax :) Well, in principle Arduino Leonardo has 2 communication interfaces, one is i2C and other is SPI, so one could implement a 2 way communication using both. In theory, one could attach one more Arduino to each of those and offload some of the tasks, but this is way out of the scope of my project. If one needs more procesing power then switching to another microcontroler is a much better idea. There are plenty of ffb wheel firmware for STM32F407, like OpenFFB, VNM, EMC or MMOS.

For start, you can use HEX fw-v220 for Leonardo as this is the most basic and easy to get setup. You only have to connect encoder to get some functionality and first signs of life.

RaceMax, 2023-08-29:
Thanks for answer) I'm only on start and so badly in theme. 

to topic: "...NRF24L01 to communicate between the wheel and the Leonardo..." - thinked about that trick, but, if removed wires from wheel, needs other microcontroller and power supply on the wheel, is in't?

jwinfieldnz, 2023-08-29:
With the SPI pins on the leonardo already populated it sounds like it is dead in the water. You wouldn't use the shift register on the nano if you were using SPI maybe? I think a curly cord is a solution. It would just be cool to have a quick release with out having to plug in a cord. First world problems:)

ranenbg, 2023-08-29:
Yes, if say I make an SPI version of button box, I need to free up pins 11, 13, 7 which means no PWM+dir mode, no FFB clip led indicator and no digital button on pin 7. It's not that much of a sacrifice, but another problem is I'm at about 95% of used memory already. Once I start adding libraries required for NRF it will probably get passed by 100%, then I would need to start cutting down other features which I'm not a fan of.

RaceMax, 2023-08-29:
Just theoretically. Replace Nano with Nano 33 IoT or some other BLE models?

ranenbg, 2023-08-29:
Yes, it's a good idea, but that's only 1 part of the story. Still, a transmitter board and ble devices (inside wheel rim) need to be coded and an appropriate communication format has to be developed, which I really do not have time or motivation to get into.

closed 2023-09-13 by ranenbg