ranenbg/Arduino-FFB-wheel #29: Encoder help

tote46, 2023-09-12:
Hi i have a problem in which my encoder only reads movement clockwise even if irl i spin it clock wise or counter clock wise , what could it be?

ranenbg, 2023-09-12:
Hi, the encoder will only turn in one direction when only one of its two channels is connected. Check your encoder wiring again and it should be fine, given that encoder itself is working ok.

tote46, 2023-09-12:
Hello I checked and it seems wiring is OK, I've tried everything: connecting a and b phase to respective pins, only works one side , inverting connections, still only works one side, connecting a phase only still the same, connecting b phase only still the same.
I have not tried using pull up resistors as I do t have 10k resistors rn, for my encoder might be a must because is an open npn collector.
Also wheelcheck and other axis showing apps don't even recognize any movement, only arduino gui recognize some move.
Also I'm having problems with other axis,  when connecting some analog Hall sensors, they work they output voltage but for example braking axis triggers Y, RX and the other axis in app.
Also even when everything is unconnected only leonardo is on, the app triggers the 20 button. When I connect hall sensor and use a magnet near it it changes the trigger from button 20 to button 22 so it seems like it's detecting it like a button?

tote46, 2023-09-12:
Could it be I'm flashing it wrong? Do I need to restart pc o do something after ploading firmware?

ranenbg, 2023-09-12:
Which firmware did you upload? I'm 99% sure you uploaded the wrong one :)

tote46, 2023-09-12:
Hi ye i uploaded the last one as i thought it was the most refined, i plan to use only steering axis and some analog axis for pedals. Which one should i use?

ranenbg, 2023-09-12:
Yes, exactly as I suspected :) You uploaded probably v223. This firmware expects an i2C communication device like a DAC or ADC, if you don't connect one then the firmware will just hang.

For your use case, I recommend vanilla firmware for Leonardo v220.

tote46, 2023-09-12:
Now I feel bad cause you're right lol you spend lots of time doing this and I want it to work right away doing nothing , sorry.
Ok gonna try that one.
I'm having problems with my bts 7960 as I cannot make it work, I have a cordless drill motor 18v and a atx psu, both have thin wires so maybe I cannot get sufficient current flow , do you know anything bout this? Idk how to test if it's a psu problem or pwm pins problem how can I check that?

ranenbg, 2023-09-13:
You opened the encoder help issue, which I think we solved now, so I'm going to close this one.

As before, first, go through all the previous issues. Many people already asked about BTS7960 wiring and troubleshooting, there I went into details on how to check things. If you still have some questions you may open a new issue. Yeah, generally you want to put the correct wires that can support 10A at least between BTS and the motor. Getting a cheap multimeter is a good start. You have to be able to measure the voltages at least.

closed 2023-09-13 by ranenbg