ranenbg/Arduino-FFB-wheel #34: Pomoc

dericakasapin, 2023-11-02:
zdravo brate, vidim da si moj sugradjanin i ako mi ti ne pomognes ne znam ko ce. U glavnom imam problem sa enkoderom i arduinom. Sve radi ali enkoder nikako ne moze. Kad stavim na pinove 2 i 3 radi ali na tx i rx nece nikako. Da li je problem u arduino ploci ili kodu. Da li je moguce da se neki hex kod izmeni da bih mogao da stavim signale na pinove 2 i 3. Ne treba mi ffb dovoljno mi je da imam volan gas i kocnicu. Ako ovo vidis mnogo bi mi znacilo da mi pomognes. Hvala u napred

ranenbg, 2023-11-03:
Cao Zemljace :) Najpre budi siguran da si uploadovao HEX v220 ispravno, mozes videti u mom wheel control programu verziju firmware-a. Sledece stavi pullup otpornike na kanale A i B od enkodera, mozes da probas od 10k do 1k bilo koji. Ne mogu da promenim HEX i prebacim enkoder na pinove 2 i 3 jer pin 3 nema hardwerski interupt, a pin 2 vec koritstin za Z kanal od enkodera.

dericakasapin, 2023-11-03:
zaboravio sam da napomenem da enkoder brojis amo u pozitivnom smeru bez obzira na koju stranu okrecem. Hvala na informacijama, probacu ovo pa se javljam!

dericakasapin, 2023-11-03:
Da li je moguce da mi je crkao neki od pinova ex tx, jer nikako nece da radi, a radi sa obicnom kodom koji sam nasao za testrianje enkodera na pinovima 2 i 3?

ranenbg, 2023-11-03:
Da, moguce je da je jedan od Tx, Rx pinova mrtav. Ako ti registrure pomeraj enkodera samo u jednom smeru, to znaci da jedan od kanala A ili B nije registrovan.

ranenbg, 2024-05-06:
Again, once more the user hardware-related problem, has nothing to do with the firmware itself, so I'm closing the issue.

closed 2024-05-06 by ranenbg