ranenbg/Arduino-FFB-wheel #35: Pwm pins issue

Hulababa-22, 2023-12-13:
Hi so i have an issue where the ffb of my wheel doesn't work if i turn my wheel clockwise. after some diagnosis i think its the pin 9 on my arduino that is broken, can you change the pwm so it uses pin 11 and 10 instead of 10 and 9? so it will be 11 for rpwm and 10 for lpwm. or add a feature to choose pwm pins?

ranenbg, 2023-12-13:
Hi, unfortunately, it's not as simple as that.

I can't change the PWM pins, because not all PWM pins are the same. These, pin9 and 10 are connected to TIMER1 in Arduino which is the 16bit timer, other pins are connected to 8bit timers. What this means is that only on pins 9 and 10 you can get a PWM signal with enough resolution and frequency that makes sense for an FFB wheel.

Just buy another Arduino :)

Hulababa-22, 2023-12-13:
oooh ok ok thanks for the answer and can i ask is it possible to run this firmware with a stm32 board?

ranenbg, 2023-12-13:
No. I didn't invest time into porting the firmware to stm32 microcontrollers, because there are already 3-4 such firmware that are available. For example openFFB, vnm, EMC or mmos.

closed 2023-12-13 by Hulababa-22