ranenbg/Arduino-FFB-wheel #36: Force feedback stops working

Matteo2607, 2023-12-14:
I encountered a problem while using the software. When I connect the Arduino and start playing everything is ok, but as soon as I hit a curb, or a sudden change of direction or in any case a rapid variation, the force feedback stops working. The peculiar thing is that only the force feedback stops working, while the encoder and pedals continue to work, but to solve the problem I have to disconnect and reconnect the Arduino, which is a problem in the race. Is there any parameter I can change to fix the problem? or some other firmware version to use?
I currently use 200a.hex
the configuration is as follows:
Bts7960
arduino leonardo
24v 500w dc motor
encoder 1024ppr

ranenbg, 2023-12-14:
Hi, that seems wierd, I didn't come across such issue. Which game is it?

Firmware v200a is not the lattest, give it a try with v220a.

Matteo2607, 2023-12-14:
the problem occurred on the racing set-up and the competition running set-up.
trying to do right left quickly in the gui I encountered the same problem.
Now I will try with the 220a. 
And I make a video with the problem.

ranenbg, 2023-12-14:
You have to be very careful, with such a powerful motor. It has a big back EMF, it can easily generate voltage above 27V which is dangerous for bts. You may want to look into some brake resistor circuits.

Matteo2607, 2023-12-14:
The encoder Is okey.
This is the video, where u can see the issues.
https://streamable.com/jlf024
https://streamable.com/q7wzlg

ranenbg, 2023-12-29:
Unfortunately, there is no information important for me to see what is the actual problem. You didn't mention the conditions of when this happens and if it is reproduceable. Try enabling ffb monitor in wheel control and follow what happens with ffb signal. See if you can pinpoint moment when the ffb cuts out.

Matteo2607, 2024-01-03:
Hi, sorry for my absence. I solved the problem by setting the ffb to approximately 50%, and the minimum force to 3%, in this way the problem does not arise.

ranenbg, 2024-01-03:
Great that you managed to solved it.

So the issue was due to your hardware, probably something was either overheating or entering into some sort of protection, overcurent or overvoltage.

ranenbg, 2024-01-03:
I'll close the issue as it was not firmware or software related.

closed 2024-01-03 by ranenbg