ranenbg/Arduino-FFB-wheel #38: sharing what I'm experiencing and some suggestions

Notriatv, 2023-12-23:
Hello friend Milos. I'm already using your GUI to control my Arduino Leonardo board, and I'm really enjoying it. Thank you for your work.

But I would like to know if the following settings are possible?: a simple AB encoder without a Z index (just rx tx ports) (which is my current use, and I don't see any issues using the GUI to manually center the steering wheel). Then, the analog pins would be used for the accelerator, clutch, handbrake and H-shifter XY on A3, A4, and A5 as I saw in your diagram.(or can we, maybe, use A3 for an optional extra axis for some randon use (like a second potentiometer brake pedal) if it´s not really necessary for reverse gear) Additionally, a load cell on pins 4-5 for the brake pedal and 9-10 for the BTS driver.

If I'm correct, there would be 8 free pins left, which could be used in a matrix 4x4 (16 buttons?) or 3x3(9 buttons?) and 2 additional inputs for the paddleshift (this would prevent false inputs if shifting gears and pressing another button at the same time.

summarizing it would be: 
1 encoder steering axis
5 axis  for accel, brake (with LC), clutch, handbrake and a optional potentiomenter extra axis (if possible)
16 buttons via matrix scheme or 9 butons plus 2 separeted inputs from this matrix
1 bts driver (pwm +- mode)
 
I find something like that in your  "v192azf" firmware where we have xy shifter option and load cell at the same firmware, but I was having some trouble with autocalib pedals.. manual calibration works fine to me :) 
so..what's do you think? is that possible mission? I follow your step by step and I'm able to compile to arduino board via IDE but I haven't knowledge yet to edit the code :(

I'm currently using v222zh.hex firmware, things are good but I have no inputs working I don't know why but they (buttons 0 to 11) starts to blinking when I turn my ffb power supply on 
again thank you so much for your work

ranenbg, 2023-12-23:
Hi, tnx for support, it means alot to me.

What you are describing should be the firmware version v222ht but without a shift register (button box). This should be doable, I'll check, there ahould be enough memory for such a combo.

In HEX v222zh, the buttons are blinking because the firmware expects a button box input. This is arduino nano with my button box firmware. If you don't connect one and leave the pin6 unconnected it's normal. It's allways good practice to gnd all unused pins.

ranenbg, 2023-12-23:
Ok, I've just checked and I have bad new.

Unfortunately, HEX which has load cell can't use XY shifter at the same time due to memory limit:
Sketch uses 29294 bytes (102%) of program storage space. Maximum is 28672 bytes.

Also, when using 4x4 button matrix, it's not possible to use at the same time XY shifter or shift register. So that is why I have not already included such a HEX combo in the release :(

Notriatv, 2023-12-26:
@ranenbg Hii, thank you for the explanation!! I see, indeed, the memory of atmega32u4 is limited, but I really like using it, and I don't see the need to use an STM instead. With your software controlling the board, I will probably stick with this platform for a good while. However, I'd still like to know if it is possible to create something like a firmware "v222f" for a pedal with a load cell and an analog H-shifter, and the unused ports could be some simple inputs as in  "v192azf"? what do you think? Also, I'm already using a second Pro Micro with Simhub to control some LEDs, motors, axis and matrix buttons..the things are fine running together, thanks !!

ranenbg, 2023-12-26:
I have to check if load cell and XY shifter can work at the same time.

Notriatv, 2023-12-26:
I see.. that's ok , you provided us enough options for vary setups, and it gave a whole new driving experience, especially in Assetto Corsa. The 'min pwm' option really helped eliminate the dead zone in the force feedback near the center and made the steering wheel much quicker to correct the car on the track! Thank you for that, and thanks for your attention. I'll try to buy you at least a coffee or a beer for this. TNX!!

closed 2023-12-31 by Notriatv