ranenbg/Arduino-FFB-wheel #39: Load Cell

EliDev05, 2023-12-28:
Hi, I'm upgrading my setup with load cell on brake pedal.
I have a 10kg load cell and HX711 module connected to Arduino Leonardo according to this schematic:

Red E+
Black E-
Green A+
White A-
GND to general GND of Arduino
VCC to 5V of Arduino
DT D4
SCK D5

I have installed fw-222ah, and the first time i ran it, the wheel control application correctly read output from the load cell, then i unplugged and re-plugged the Arduino and the load cell stopped working... now the Z Axis is blocked on 1.0 value.

I really don't know how to fix this, I'm worried that my load cell or HX711 fried up.
Thanks in advance if somebody can help me.

ranenbg, 2023-12-29:
With the information you provided it's hard to say where the problem is. You wired it correctly and the fact that it was working well the first time you run wheel control confirms it.

There is low chance that HX711 or load cell died just like that. It's probably some lose connection, some wire not holding properly, check that DT and CLK wires are connected firmly in pins 4 and 5. Even without load cell connected, when touching HX711 pins by finger you should see some noise readings. Make sure that you do not reverse A+, A- wires of load cell, as negative values are not displayed in Y axis, only positive. When it shows a -1 reading on Y axis, this is usualy sign that there is communication with HX711, or that it's interrupted.

In wheel control, there is a slider to adjust brake pressure, by default it's at 45 (out of 255). Try changing it a little back and forth. Each time you change it, a tare of load cell (zeroing) is performed. The most sensitive (1:1 reading) is with value of 4.

Hope this helps. Good luck.

EliDev05, 2023-12-29:
I think that there are some software issues, or maybe the HX711 module goes on some stand-by mode...
First i have switched the HX711 module on 80hz mode and i have discovered something new:
my pedals set are connected with rj45 cable to a custom arduino shield, so i can connect and disconnect all the pedals with just one cable...
when i connect the pedals cable, for one second i can see the brake axis (the yellow one) at some values and if i give some pressure to the brake pedal the value increase, then the value drops to 0 and the Arduino can't read any value from the brake pedal.
Without powering off the Arduino, if i disconnect and reconnect the pedals cable the same thing happens.

I have attached a video where you can see this strange behavior:
https://github.com/ranenbg/Arduino-FFB-wheel/assets/59321870/5ef2ec64-843a-497d-aaab-e1222900b1ad

Thanks for the help.

ranenbg, 2023-12-29:
Yeah, this seems very wierd. Not sure what you can do here other than playing with max brake pressure slider. You had it at 45 or 40, this is quite high value for such a tiny 10kg  load cell. I have a 75kg one and I use 45 for it as this feels best for me. You can try something lower, for example 32 or less.

EliDev05, 2023-12-29:
Thanks i will give a try with a lower max brake pressure and with the last version of wheel control... If all of these solution won't work i will buy a new load cell with higher kg like yours, maybe 10kg is to low.
By the way the ground is the common for the shield and all devices and component so i don't think that can cause this problem.

I will keep you updated, thanks a lot for the help.

EliDev05, 2023-12-29:
I have tried to lower the max brake pressure value, update to the last version but nothing helped.
I'm not sure but maybe there are some interference/false contact in the rj45 connector...
I will buy a "larger" load cell and a different HX711 module.

ranenbg, 2024-05-06:
I'm closing this issue as it was related to some hardware issue of the user and it's not a firmware-related problem.

closed 2024-05-06 by ranenbg