ranenbg/Arduino-FFB-wheel #41: MIcro and XY shifter

jwinfieldnz, 2024-01-01:
I don't see any specific wiring for the micro and XY shift (f) firmware. So I have connected it up to A2 and A3 and this seems to translate well to RX and RY but the XY Shifter only sees movement in a diagonal direction when moving RY. If you see the diagram below RX moves but there is no movement in the XY shifter at all. 

[image]

Just wondering if it is fully implemented in MIcro or not. I think I have tried all F firmware.

ranenbg, 2024-01-01:
I didn't put any specific wiring for XY shifter for micro HEX, because it is the same as for leonardo HEX. You have to use pins A4, A5 (x, y) for analog shifter.

What you observe is not an issue at all. You left A4, A5 unconnected (floating), while you connected pedal axis pins A2, A3. Which ever voltage was on pin A3 was reflected with quite some noise to floating pins A4, A5 in the same way. That is why you saw a diagonal movement, as both pins received the same input.

Wire it correctly and it'll work just fine.

jwinfieldnz, 2024-01-01:
Ahh TIL there is a Micro and Pro Micro. Strangely the pro micro (the one I am using) skips A4 and A5! Thanks for your reply

closed 2024-01-01 by jwinfieldnz

ranenbg, 2024-01-01:
You're welcome.

You are right, it's somewhat missleading. HEX for leonardo and micro is the same, as both boards have the same pins. Only proMicro board has a different format and I'm using special HEX for it with replacement pins, but I named it micro HEX. It is a bit confusing, I admit, however, I explained that clearly in the first paragraph of readme.

jwinfieldnz, 2024-03-11:
This is the first image on the readme, it doesn't seem to mention the alternative pins for the XY shifter and micro. I have figured that out. What would be the reverse button input?

[image]

Would it be A0 or A1?

[image]

jwinfieldnz, 2024-03-12:
Ahh pin 4.

ranenbg, 2024-03-12:
Hi, yes, as you can see in both pictures I wrote button0 is on pin 4, or pin A3.
The bottom picture shows a h-shifter wiring and there you can see in more details, which of these two pins is used for reversed gear, depending on HEX version that you use. If you use HEX with a load cell v222m, then one analog pin A3 gets free and I used it for button0. Otherwise, it's on pin 4, for HEX v220m, or v220.

ranenbg, 2024-03-13:
Ok, good, I'm closing the issue as there was none. Please pay more attention to the documentation I provided and we can avoid these kinds of questions.