ranenbg/Arduino-FFB-wheel #43: How to get 32 buttons to work?

johanneslerch, 2024-02-01:
Hi there, first of all, thank you for sharing this awesome project.

I'm trying to get 32 buttons to work (I'm using four 8-bit shift registers chained together). By using DEBUG_SERIAL I was able to confirm that I can read all 32 buttons values, but it seems only 24 buttons receive push events in windows.
I'm looking at the USB game controllers window, which successfully lists 32 buttons in theory, but the first 8 will never be pushed while I can successfully use the latter 24.

Here is a screenshot of what I'm looking at (note that all buttons have pull-up resistors and I didn't negate any values yet, which causes them all to appear pushed while they are actually not):
[image]

My code changes can be found here: https://github.com/johanneslerch/Arduino-FFB-wheel/commit/8112df4e093598882267c2dc1da1f56ec0bb7158

I assume I'm missing some change in the USB device setup. Any hints would be appreciated!

ranenbg, 2024-02-01:
Hi, tnx. Congrats on what you did so far, very nice work. It's important to understand there are a few ways to do this.

Have a look at hid.cpp this is the usb device setup part that you are probably missing, as I didn't see you made any changes there. There is a part in hid descriptor which defines a hat switch and buttons come after it. There are 8 bits reserved for hat switch that come first, while last remaining bits are for buttons.

Input bytes function "ver 5" that you use at the moment probably has to be changed as well.

johanneslerch, 2024-02-02:
> Have a look at hid.cpp this is the usb device setup part that you are probably missing, as I didn't see you made any changes there. There is a part in hid descriptor - input report definitions, which at the end defines a hat switch and buttons after it. There are 8 bits reserved for hat switch that come first, while last remaining bits are for buttons. This structure must be respected.

Do you refer to this section in hid.cpp ? That's at least the only one that I can spot mentioning a HAT SWITCH and Buttons. Note that I did not change hid.cpp directly, but indirectly as I changed the `#define NB_BUTTONS 24` to 32 in USBDesc.h.
```
  0x09, 0x39,                     // USAGE (HAT SWITCH)
  0x15, 0x01,                     // LOGICAL_MINIMUM (1)
  0x25, 0x08,                     // LOGICAL_MAXIMUM (8)
  0x35, 0x00,                     // PHYSICAL_MINIMUM (0)
  0x46, 0x3B, 0x01,               // PHYSICAL_MAXIMUM (315)
  0x65, 0x14,                     // UNIT (Eng Rot:Angular Pos)
  0x55, 0x00,                     // UNIT_EXPONENT (0)
  0x75, 0x04,                     // REPORT_SIZE (4)
  0x95, 0x01,                     // REPORT_COUNT (1)
  0x81, 0x02,                     // Input (Data,Var,Abs)

  0x05, 0x09,                     // USAGE_PAGE (Button)
  0x15, 0x00,                     // LOGICAL_MINIMUM (0)
  0x25, 0x01,                     // LOGICAL_MAXIMUM (1)
  0x55, 0x00,                     // UNIT_EXPONENT (0)
  0x65, 0x00,                     // UNIT (None)
  0x19, 0x01,					            // USAGE_MINIMUM (button 1)
  0x29, NB_BUTTONS,               // USAGE_MAXIMUM (button NB_BUTTONS)
  0x75, 0x01,                     // REPORT_SIZE (1)
  0x95, NB_BUTTONS,			          // REPORT_COUNT (NB_BUTTONS)
  0x81, 0x02,                     // Input (Data,Var,Abs)
```
I believe this change resulted in USB GameController dialog showing 32 buttons instead of 24. Is there something else that I need to take care of?

> Input bytes function "ver 5" that you use at the moment probably has to be changed as well if you want to keep the hat switch. 

I didn't plan to use any hat switch. Best to my understanding, the hat switch is sent as part of the "buttons" variable, right?
I've dropped the left shift to avoid messing with my 32 values in the buttons variable: https://github.com/johanneslerch/Arduino-FFB-wheel/blob/8112df4e093598882267c2dc1da1f56ec0bb7158/brWheel_my/Inputs.ino#L337
How would I disable the HAT SWITCH completely? 

>Do not forget to copy changed hid.cpp file as well as all other usb core files you changed in correct folder, otherwise it'll use the old ones even after making changes to them in arduino ide.

Phew, this I completely missed. Wondering why it seems some changes did have an effect. I will double check this one.

ranenbg, 2024-02-02:
Yes, this is part of hid decriptor I wanted you to have a look at. Yeah, I also saw this wierd N+1 button showing up. It seems that the number is correct only when one uses 16, 32 or 64. Anything in between gets the wierd +1, but this is only a small cosmetic issue.

To completely remove hat switch you can just comment out its usage page and everything below it,  
just up to the buttons usage page, but I do not recommend it. You may get stuck in damaged hid descriptor and the device will not even show up, it's extremely hard to debug this, you must know exactly what you are doing. Yes, out of 32bits sent there are some reserved for hat switch, that come first.

Instead, I would rewrite the input report function to have 8+32 buttons, something like 16,16,12,12,12,8,32. But I leave it up to you to decide. If you remove hat switch then you can utilize all 32bits for 32 buttons, using the ver 5 function.

johanneslerch, 2024-02-02:
Got it working now. Of course I was missing something obvious...
I did not uncommend the declaration section for axis rz, but started using ver5 which wants to send rz. That means, I ended up sending rz, but it was interpreted according to the declaration, which means my empty value for rz was interpreted as value for the first buttons.

My current state is here: https://github.com/johanneslerch/Arduino-FFB-wheel/commit/aa2016fdbf74139757848daa13cfe0c217a8b9e1

In a next step, I want to clean that up and create a ver6 as you suggested using two 32bit variables allowing to push 4 bits for the hat switch + 60 buttons. That should be enough, even for me ;-)

Thank you so much for the support.

I believe, it may be time to share a little about my other plans (as I may run into more questions in the future).
I'm planning on 
* 32 push buttons, 
* a rotary encoder on pins D4 and D5, which I will transform into two buttons that will be virtually pushed depending on the direction of rotation
* a 3-axis joystick with an integrated push button on A3, A4, A5 and D12. 
* a possible further extension of buttons (if I ever get to add buttons directly to the wheel) via a Nano in the wheel and UART communication via D2 and D3

If from these, you already see me running into a problem, let me know. Though, the custom PCB is produced and now successfully tested to be working already, so I'm kind of committed to make it happen ;-)

closed 2024-02-02 by johanneslerch

ranenbg, 2024-02-04:
I see, tnx for all this info, nice :)

I never tried or even thought to stack arduino nano (software shift register) with an actual hardware shift register IC, that is why I say it can't work at the same time. But maybe it can work without any additional code modification, you tell me. In principle nano is simulating the operation of shift register, so as long as everything is in sync concerning the PL (pull) signal logic, it should be fine in theory.

johanneslerch, 2024-02-08:
Ah, there is a misunderstanding. I did plan to add the nano to different pins than the shift registers.

ranenbg, 2024-02-08:
Oh, I see. Yeah, that may work I guess.

Originaly in BR wheel code there were two pins planned for such purpose, one for h-shifter and another for wheel rim, but it was never implemented. I ditched it and just exanded/restructured a single shift register readout with arbitrary bits number.

MrCjxer, 2024-07-19:
Hi @johanneslerch 
did you make it success ?

johanneslerch, 2024-07-19:
> Hi @johanneslerch did you make it success ?

Hey, the answer depends on what you are refering to.
The part with the nano for additional buttons within the wheel rim is something I didn't even start yet.

All the parts shown in the CAD picture above are realized and fully functional.

johanneslerch, 2024-07-19:
Found your issue and therefore, assume it's about how to get the 32 buttons to work using shift registers. I will continue to describe in this issue to have all details of my solution in here.

I've been using SN74HC165N shift registers. I strongly recommend reading about how shift registers work. You will need to understand what CLOCK, LOAD, and SERIAL pins mean, because these you will need to your Arduino Leonardo or whatever board you are using.

I'm attaching you some schematic of the wiring that I did:
[image]

Then I did some changes to the code that you can find on my fork https://github.com/johanneslerch/Arduino-FFB-wheel/commits/master/.
The respective commit is this one, though, I'm not entirely sure anymore if a later one fixed something or it worked in that state already: https://github.com/johanneslerch/Arduino-FFB-wheel/commit/aa2016fdbf74139757848daa13cfe0c217a8b9e1

ranenbg, 2024-07-19:
Tnx very much for this info, it's a super useful resource :)

MrCjxer, 2024-07-20:
> Hey, the answer depends on what you are refering to. The part with the nano for additional buttons within the wheel rim is something I didn't even start yet.
> 
> All the parts shown in the CAD picture above are realized and fully functional.

hi again
Exactly what you said 👍 :)
I need more buttons like you

=====================
woooow . soooo nice and thank you
very useful

> 
> Later commits extend the HID to support up to 64 buttons as I ended up with 32 buttons via shift register + some more directly connected to the Leonardo board.

1- but you mean its support 32 Button for now? and maybe someday you will edit it and extend the HID to support up to 64 buttons?
 

> some more directly connected to the Leonardo board

2 - how many buttons directly connected to the Leonardo board?
because i have the same board (Leonardo)

> Then I did some changes to the code that you can find on my fork https://github.com/johanneslerch/Arduino-FFB-wheel/commits/master/

3 - for this changes that you did, need any change on FFB-gui?
https://github.com/ranenbg/Arduino-FFB-gui

> I'm attaching you some schematic of the wiring that I did

4 - on the schematic, what is " WHEEL-COMM " and " TEMPOMAT_ENCODER "?

sooooo sorry for many questions, I'm noob :)