ranenbg/Arduino-FFB-wheel #44: ffb max output force?

Notriatv, 2024-02-07:
Hello! I wonder if there is a possibility of adding this feature? Because in some cases FFB is too strong, and the power supply ends up shutting off.(mostly on beamng or iracing colisions) I know that if I reduce the overall gain, I can solve this, but I don't want to lose the minimum forces.. what do you think?
Thanks in advance :)

ranenbg, 2024-02-07:
Hi, this is not a real issue of the firmware. You can't have both at the same time, there is allways some tradeoff. What you can do is use lower voltage for your motor when powering BTS7960, if you have a psu that has voltage output adjustment.

With firmware you can play with general gain and min pwm sliders. You can also try out different PWM frequencies. Remeber that lower PWM freq will give you more ffb resolution as PWM signal will have more steps inside one period.

Notriatv, 2024-02-08:
I see, for sure that's not an issue but I don't know another way to contact developers here on github lol
I was just thinking about that because I saw this feature or something like that on EMC and AFFBWheelGUI (not tested here yet) in "cut force" description but I'm completely noob and don't know how you guys do the magic :)
ok I'll keep playing with sliders to find perfect sets for every game..
I'm using a 200w dc spindle for CNC with 1:10 belt reduction and a 24v 500w(not real) psu
I once tried running it with a 24v battery and ffb was great without turning off due to "ffb peaks"
I'm currently running with 8khz for racing games since the engine sounds are louder than pwm noise and 16khz for ets2

Notriatv, 2024-02-08:
oh yess another quick question.. is it possible to use some BL motors with your firmware? like those we find on hoverboards

ranenbg, 2024-02-08:
Just as I thought, 24V 500W motor is scary :)
EMC and AFFB have 8bit ffb with only a few effects, while mine is 16bit will full ffb effect support which takes a lot of memory, so that is why they were able to fit ffb filtering on it. I was once a complete noob with arduino programing and ffb, but I'm working on this since around 2017 and had a strong motivation to learn more. I only used sources available online.

In the game you can always set ffb strenght in such a way that clipping does this job of cutting high peaks for you. If you set it on a higher side, such that there is no clipping under normal conditions, so ffb signal is most of the time close to 100%. When a sharp peak comes, it will be just clipped out as game can't send more than 100% ffb. You can use my realtime FFB monitor in GUI just click on a square button next to a general gain slider and run the game in window mode such that FFB graph is just behind and below game window.

Howerboard motors, yes, for that reason I made a special pwm mode I called RCM. This is the same as the one used in RC cars for servo motors that you put on the front wheel for steering. Howerboard electronics can work with such pwm input signal. There were a few guys asking around about this and I made it upon their request. I'm not sure of they got it to work, they said there were problems with it concerning calibration of pwm input on the howerboard electronics side, there is some knock-off firmware for this board somewhere.

Notriatv, 2024-02-08:
oh I see.. big thx for sharing your knowlege with rest of us 👍
 for sure one day I'll try bldc just to satisfy my curiosity :) and learn to use that cliping window
if you want to see something of my project that's my yt channel https://youtube.com/@NotriaTV?si=KOfzXNBMi3usPVOP
thx again!!

closed 2024-02-11 by Notriatv