ranenbg/Arduino-FFB-wheel #47: Invert X or Y manual shifter

jwinfieldnz, 2024-02-21:
I had a quick look through the code but I couldn't find a flag to invert an axis using the `f` firmware. I am using a PS3 analogue stick and I can't really swap the polarity as it's mounted to a PCB. The issue is that left and right are ok but pushing forward and back are inverted. I guess it doesn't really matter in a game but it would good to see the right behaviour in the wheel_control tool.

ranenbg, 2024-02-21:
Hi, there is no such flag, I didn't implement it in firmware. All axis reversal and ffb reversal stuff I deliberately left out as they can be done by hardware with correct wiring. First reason is that this uses extra memory space, I was trying to save and second is that it will create a mess in diagnosing what is reversed and where.

However, the suggestion is resonable and I may have to revisit it at some point, but I'm not a fan of this idea, as I made decision not to do it from the very early days of firmware developement. The firmware is ment for DIY community, where one has the skills and freedom to do his own wiring.

In your case, there are two options: cut some traces on the pcb, or just put one extra line of code after analogRead to reverse the readout like: value = 1023 - value;

jwinfieldnz, 2024-02-21:
So something like `shifterX = 1023 - shifterX;`? It would be great to have a release with this feature, particularly if you can set the axis to invert in the wheel_control tool. Agreed, most will use individual pots and not face the issue at all. These HW-504 however are cheap and hardy and are ideal for this application other than this one small issue.

ranenbg, 2024-02-21:
Yes, just like that, but place it right after shifterX=analogRead().

I'm sure that you realize how much work it is to make a new release of both firmware (20+ HEX files) and wheel control, just to add one tiny feature that almost no one would use. One line of code change in firmware is not worth a new release. I have however a few things on a to do list already and thinking how to implement all of them without braking backwards compatibility is not trivial at all. This is all given that there is enough memory left in arduino.

Even with reversed Y axis, the H-shifter will be perfectly functional. Only the reversed gear will be in 2nd instead of 6th or 8th, which is not that big of a deal.

jwinfieldnz, 2024-02-22:
Ok no worries. I will try and find a workaround on the HW-504.

ranenbg, 2024-05-06:
Since I have released a firmware version that supports advanced shifter config, including the inverting both X and Y shifter axis, I'll close this issue.

closed 2024-05-06 by ranenbg