ranenbg/Arduino-FFB-wheel #48: Firmware for STM32F103C8T6 (STM32 Blue Pill)

chizhkovanton, 2024-02-28:
Hello! I would like to know if there will be firmware for the STM32F103C8T6 (STM32 Blue Pill) board? Do we need to change the for memory? Or what will be the pinout for this board with Arduino firmware?

ranenbg, 2024-02-28:
Hi, I have no plans of porting the firmware to any other chip than AtMega32U4. That would require writting everything up from scratch pretty much. You can't just use a different MCU like that and especialy not STM32 that has a totaly different architecture.

I mean, you can try and see what happens, but I'm sure there will be a ton of unfiable compiler errors, that it makes not much sense to port it.

chizhkovanton, 2024-03-06:
Okay, anyway, thank you so much for your hard work. But there is a question: what if the Wheel Config writes the firmware version as 0.0?

ranenbg, 2024-03-07:
Tnx, you're welcome.

Wheel Config is a part of AI wave firmware and it is not compatible with my firmware at all. It is true that my firmware is based on some very early version of it, so as ESP Wheel, so as BR wheel, but that's where all similarities end. My firmware is a huge step up from those in almost every aspect, concerning FFB quality and other features.

If you want to do any configuration of my firmware, you have to use the wheel control GUI that I made, it is in my other repository on Github. Alternately, you don't even have to use my GUI, I have provided full documentation in RS232_commands.txt so you can use any program capable of serial communication to configure it. My firmware is also presented as a virtual serial port to any windows PC, so it's easy to work with it and configure it once you know which command to send.

closed 2024-09-09 by ranenbg