ranenbg/Arduino-FFB-wheel #49: Wiring help (optical encoder)

guilhermevleite, 2024-03-21:
Hi, I am sorry to create an issue just for this, but I can't figure this out.
So, I got an optical encoder that needs between 5V and 25V to power up, and the arduino cannot feed this much. My solution was to get an external power supply and a tension regulator to bring the power from 110V -> 15V -> 5V, and then power the encoder.
That being said, I still cannot get the wheel_control to register the encoder signals. I know the encoder is working because I tested it with a simple sketch.
What am I doing wrong? In the image is my wiring setup, A0-A3 are shorted to GND.
Is there a discord server I can also try to get help?
[image]

ranenbg, 2024-03-21:
Hi, the wiring diagram you presented is correct, this should work.

Which firmware version did you upload to your proMicro? It could be that you uploaded hex ending with 3, like v223. That will not work without an i2C dac connected, so make sure you have correct HEX, like v220 for beginning.

If that doesn't help, you could try with a higher voltage, for example just put a different voltage regulator like LM7812 or LM7815. If you do it, be very careful not to keep pullup resistors still connected to this voltage as you will burn arduino. Instead, connect them to arduino 5V pin, such that they pull encoder signal lines HIGH to 5V. Value of 4.5k seems a bit odd, you can go down to about 1k, but I recommend to use 3.3k and only go lower if you see encoder step skipping (drift of wheel center over time).

I have no specific discord server for my firmware, I preffer to keep the communication via github only.

guilhermevleite, 2024-03-21:
Hi, thanks for the quick answer.
I have tried a few of the hexes, right now I am using brWheel_my.ino.micro_v220cm.hex
You are correct, 4.5k is weird, I am using 4.7k, do you think I should go lower?
I will try another regulator and move the pull-up resistors to VCC, as soon as I buy the proper regulator.
Thank you again!

ranenbg, 2024-03-21:
You're welcome.

In principle, you don't need lower pullups than 4.7k. This depends on PPR of your encoder. You should be fine with this values up to 2k PPR encoders.

ranenbg, 2024-03-22:
After a second look at your wiring diagram, I noticed that you didn't connect encoder GND to arduino GND pin. This is crucial and it's definately a reason why you had no signals from encoder. Once you make this connection I'm 99.9999% sure that it will work.

guilhermevleite, 2024-03-22:
> After a second look at your wiring diagram, I noticed that you didn't connect encoder (and psu) GND to arduino GND pin. This is absolutely crucial and it's definately a reason why you had no signals from encoder. Once you add this connection to your circuit I'm 99.9999% sure that it will work fine.

THAT WAS IT! It worked!
The odometry is a little off, when I turn counter-clock it looses the center point. But that may be some loose wire in my part, will test more.

Thank you again!

ranenbg, 2024-03-22:
Make sure to enter the correct CPR value for your system. If you have encoder mounted directly on wheel shaft then CPR=PPR×4. If you have it via pulley or gear system then you have to multiply with a gear/pulley ratio.

You may also want to try with a lower pullup values like 1k, to get more fast signal response (faster pullup to 5V).

closed 2024-05-06 by ranenbg