ranenbg/Arduino-FFB-wheel #51: Kompilovana verzija ne radi

Azur12345, 2024-05-03:
Zdravo! Ja pravim volan sa ovime ali nemam enkoder. Imam samo ovaj potenciometar. A znam poprilicno o Arduino tako da sam htio samo malo da prepravim kod da radi sa ponteciometrom. Nego, prije nego sto sam poceo da modifikujem kod samo sam ga istestirao sa gasom i kocnicom. I dalo mi je dosta gresaka. Uradio sam sve kako je Readme rekao. Zamijenio sam arduino core, downgradevao boards manager I nista. Koristim Arduino Leonardo i stvarno sam zapeo. I kad sam probao hex verziju samo onako kad sam probao da udjem u Assetto corsu samo da vidim da li radi, samo se zacrnio ekran i crashovalo se. Sa ovime se vec satima zezam. Probao sve zivo. Includovao inputse i jos tona stvari ali apsolutno nista nije radilo. Za Assetto corsu probao sam da koristim Joystick library od MHeironimus-a i tad je Assetto corsa radila normalno.

ranenbg, 2024-05-04:
Hi, for the sake of others, I'll repply in english.

Briefly, this user is asking about how to compile the source code and reporting the issue with compiled HEX files, simply stating that it doesn't work in assetto corsa.

With the details you provided in this issue, there is not enough information for me to give you exact solution. You have to write in mode details (prefferably in english) all the steps you took so far and which kind of errors you get from arduino ide, which ide version do you use, from which arduino boards version you downgraded, which HEX versions did you try so far, which wheel control version did you use.

Usual mistake when it comes to HEX files is uploading a wrong version for the hardware you have connected to arduino, or arduino board itself. Fail safe versions are the ones ending with a 0, like v230, or v230x. They do not require anything to be connected to arduino and should work without any configuration straight away. If you have a proMicro use the HEX files with "m" option. I recently released new firmware and GUI, I strongly recommend to use them. In your case, with v230x you can use a pot for steering and working FFB effects on analog axis.

Azur12345, 2024-05-05:
Sorry for the late reply and for being quite vague. Okay so, the main problem is that I wanted to use the .ino version because I wanted to tinker with it and see how it worked. So now I redid the steps that were mentioned in the readme and now it won't get past the first line of code.
The error is;
:\Users\User\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.5\cores\arduino/new.h:3:10: fatal error: new: No such file or directory
 #include "new"
          ^~~~~
And for the assetto corsa black screen/crash this is it:
[image]
I tried to do it with the simpler library(ArduinoJoystickWithFFBLibrary from YukMingLaw) and still it is giving me the same error on crash. When I tried using the standard Joystick library(from MHeironimus) it worked and launched the game without a problem. It's not a problem with assetto corsa I reinstalled it and verified files removed mods reinstalled csp everything and yet nothing worked. Maybe it is a problem with my windows. I looked directly into the .h and the .cpp files and they were exactly the same as in the other one. I was using the v230x hex version.

ranenbg, 2024-05-05:
To compile the source, I'm using arduino IDE version 1.8.19 and boards version 1.6.21, try with that. The error you had says you have no new.h header file, so probably something wrong with either boards or IDE version you used, can't say more than this.

About HEX v230x, once you upload it to Leonardo, did you try to open wheel control 2.6.0? Can you send a screen shot of it, if you can get passed the configuration?

Azur12345, 2024-05-05:
I miracoulsly managed to fix the issue with assetto corsa. Turns out my dinput and other hid dll system files on windows were corrupted. Replaced them and now it works flawlessly. I am using arduino 2.3.2 and boards 1.8.5. I will try to use the approapriate windows and board version. The computer does recognize the leonardo as the controller. I will get back to you with more details.

Azur12345, 2024-05-05:
Okay now I installed arduino 1.8.19 and 1.16.21 and now it is giving me this error code:
exec: "C:\\Users\\Azur\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++": file does not exist
Error compiling for board Arduino Leonardo.

ranenbg, 2024-05-06:
Ok, good, at least some progress with HEX.

It seems that you did not copy the usb core files correctly, or that IDE is using them from a wrong location. By now, you had installed many versions and I have no idea what is happening on your PC anymore, this is going to be a pain to diagnose. That IDE v2 may have still left some of its traces. I've never seen a compiler error that serial.begin() or other ones connected to serial com functions are not working.

Anyway, it took me about 6 months to get this to compile error free, when I was just starting with Fernando's BR wheel project back in 2017, so don't give up just yet.

Azur12345, 2024-05-06:
I somehow got it working now. No erorrs anymore! all of a sudden it started working. I changed nothing and it compiled.

Azur12345, 2024-05-06:
I reinstalled the same core hundred times and it somehow started working.  Thank you kind sir.

Azur12345, 2024-05-06:
if it all of a sudden breaks I will notify!

closed 2024-05-06 by Azur12345

ranenbg, 2024-05-06:
Great, congrats and welcome to the club :)