ranenbg/Arduino-FFB-wheel #52: FFB Wheel

mustafabasboluk, 2024-05-15:
I have 1 BTS7960 driver
1 piece 1000 pulse encoder
1 Arduino Leonardo
I have 1 RS775 DC motor.
Which software should I use?
I have tried a few but FFB does not work.

ranenbg, 2024-05-15:
Hi, most probably you didn't connect something correctly. You have to check step by step each connection. Then troubleshoot, if encoder is working, if you have connected and powered bts, if bts is working, only then connect a motor. You can use a voltmeter to measure if bts gives any output.

For fail safe, use HEX firmware v230x from leonardo folder and you should use wheel control v2.6.0.

From the info you provided, you basicaly only said that you have those components, but didn't say anything how you connected them. At this point I can't help without knowing more details about your particulair wiring or component issue.

mustafabasboluk, 2024-05-15:
[image]

This is my connection.
There's resistance, I just didn't show it.
FFB doesn't work even though I installed 230x software.
When I give external power to the motor, the motor works, the motor is not broken.
When I connect it to the driver, it does not work, I could not understand my driver is faulty.
Is there a mistake in the connection?
I check it from wheelcheck with this connection and it does not work.

Translated with DeepL.com (free version)

ranenbg, 2024-05-15:
Ok, good, this is something we can work with.

Wiring is fine mostly, except for the encoder connection. This way you are shorting encoder outputs to 5V all the time, it will not give any movement and therefore, you will not see any FFB.

What you need to do is add pullup resistors instead of a red wire that shorts A, B to 5V. The resistors can be anything between 1k and 10k.

mustafabasboluk, 2024-05-15:
The resistors will be at 5v coming to A and B of the encoder, right?

ranenbg, 2024-05-15:
Yes, one end of each resistor go to 5V and other end go to A, B, Z. One pullup resistor for each encoder channel.

MrCjxer, 2024-07-14:
Hi , can i use this sensor ?
It has GND , VCC , RX , TX
need any Z phase or Z button or Z trick ?

Thank you

[image]

ranenbg, 2024-07-14:
This one I think you could use, it should be just like any other optical quadrature encoder with 2 channels, where RX and TX are channels A and B. You have to check if it can operate at 5V and if signal level at it's RX, TX outputs is also 5V. You should be able to check this with a  multimeter, but scope would be ideal. It looks like it has quite high PPR, so you can even mount it directly at the wheel shaft, it should provide enough resolution in steering axis. You will have to use trial and error to determine the correct CPR. I expect something around 3600 CPR.

For the z-index, it probably doesn't have any z channel, so you can just click center button after every arduino powerup, or connect tge button and use firmware with hardware recenter feature, this is HEX with "c" option.

MrCjxer, 2024-07-14:
hi again
yeah it work with another FW but with your FW i did not test yet , so i will test it
i think its a 2150 CPR.

i have another question for you
i want use ALL 24 button with D-PAD , should i use which FW ?

[image]

and which PINs on Arduino Leonardo should i use ?
Shift register ? or matrix buttons ? or something else ?
and which model of shift register ?

i'm sorry for my bad English bro 😅
thank you

ranenbg, 2024-07-14:
About encoder, it should work with my firmware, just do not forget to put 4.7k pullup resistors. Anything between 1k and 10k will do, but lower is better as it has a faster signal response and is less prone to step skipping.

For buttons, you can use max 16 with separate arduino nano to which you upload my button box firmware. Other 8 buttons are reserved for analog H-shifter. It is possible to use all buttons available, but that requires modifying the source code, it's a relatively easy task if you know exactly what you're doing.

closed 2024-09-09 by ranenbg