ranenbg/Arduino-FFB-wheel #53: Deadzone in inputs?

RandomHuman97, 2024-06-12:
When I try to turn the wheel, there is a very noticeable dead zone (the encoder moves and pulses, but the wheel doesn't send it as an output), is there any way to disable/lessen it?

ranenbg, 2024-06-12:
Hi, you are probably refering to the FFB deadzone around wheel center when autocenter spring force is active.

The cause of this issue is in the firmware or arduino, but inthe way how a DC motor operates. It simply needs some minimal voltage/current to make it start spinning.

In my firmware I made a setting to compensate for this motor startup current. You will see a slider in wheel control GUI named "min pwm torque", which does exactly that. Its available range is from 0% to 20% and you will have to play with it to find a sweet spot. Start with about 5% and go higher from there. Whay you want is to minimize the ffb deadzone around center, but prevent any small oscilations, so it's a trade off. You can adjust it with 0.1% increments to really fine tune it, but be aware that this will also depend on pwm frequency and general gain slider value, so make sure you have set those two first.

RandomHuman97, 2024-06-12:
Nevermind, I was talking about the rotary encoder deadzone, but turns out it was a software issue.

closed 2024-06-12 by RandomHuman97

ranenbg, 2024-06-12:
There is no software or firmware issue with the encoder. You probably had a hardware issue with incorrect wiring. Make sure to use pullup resistors and to connect them correctly. I've provided a wiring diagram.

Next time if you post an issue, please put some more effort to precissely describe the problem, as we can get to the solution much faster and help others solve their similair problems.