ranenbg/Arduino-FFB-wheel #54: pmw + dip problem

Envi4, 2024-06-16:
Hi, before that, there was an affb firmware. I decided to switch to your new one, but I have a problem with PWM + dim. I only have single-way recoil, but everything works fine with PWM, just like it does on the affb. Everything works well with PWM + DIP.

ranenbg, 2024-06-16:
Hi, I'm not sure which problem exactly are you trying to describe. You are talking about pwm+dir mode. Double check if you are taking the direction signal from a correct pin on arduino, as it differs for the 3 supported arduino boards.

Envi4, 2024-06-16:
PWM+DIR Yes, I'm referring to this mode, but I couldn't get the circuit to work for me. Maybe I connected it incorrectly.

ranenbg, 2024-06-16:
Yes, that's the first thing you have to check. If you have a multimeter it's very eqsy to mesure dir signals, it's either 0V or 5V, it ahould alternate these two values when ffb changes direction.

Envi4, 2024-06-16:
[image]

I have correctly connected everything to the bts7960 itself?

ranenbg, 2024-06-16:
You are using wrong mode, as bts7960 is a H-bridge type of motor driver that needs 2 channel pwm+- mode. You can set 8kHz, it should be fine. Pins 9 and 10 are the pwm outputs, also make sure to connect 5V (vcc) to pwm enable pins.

Envi4, 2024-06-16:
Thank you very much, but what kind of board is generally used for pwm + dir

ranenbg, 2024-06-16:
There are some H-bridge DC motor drivers that use pwm+dir as input, but they are rare. Most of the time, this kind of input is used on 3 phase AC servo motor drivers, or on some 3 phase BLDC drivers. Howerboard motor driver is another example as well.

closed 2024-06-16 by Envi4