ranenbg/Arduino-FFB-wheel #56: Problem with the pedals

MatteoDav2607, 2024-07-04:
I only have 2 pedals and the remaining 2 free axles flicker continuously. How can I disable them?

ranenbg, 2024-07-04:
Hi, you can not disable the unused axis, but you can still prevent flickering. This is noise on the floating unconnected inputs and it's normal.

You should connect all unused arduino analog inputs to gnd (0V). Then use manual calibration for pedals in GUI. You just need to raise the lower callibration limit of each unused axis until it is just slightly above the zero level, if it's not already at 0 and staying there.

MatteoDav2607, 2024-07-04:
I did this and solved the problem with the unused axes, but the 2 that are needed continue to flicker. what could I do?

ranenbg, 2024-07-05:
This is quite tricky to solve comoletely, as there are several sources of noise in analog inputs:

[1] Analog to digital (ADC) conversion noise in arduino, that depends on the ADC settings. I'm using default settings, so +-1 step out of 1023, is normal noise.

[2] Unshielded cables or dirty pots. It is recommended to use shielded cables for all pots. You can use an usb cable wire, it has an outer shield plus 4 more wires inside, so you can use one such cable for up to 3 pots, where shield is gnd. For dirty pots I just spray WD40 which helps to get all the dirt out and sligthly lubricate it. This source of noise can be up to about +-10steps out of 1023.

[3] Electromagnetic interferrence - if you have unshielded wires that cary high amps with a pulsing or AC voltage close to your sensitive analog signal wires, this can create a lot of noise +-50 steps or more. Good practices are to twist motor wires, and move away any wires with 220V 50Hz AC. Shielding and earthing all sensitive analog parts, including enclosing arduino in a metal box.

closed 2024-07-11 by MatteoDav2607