ranenbg/Arduino-FFB-wheel #57: Use 24 Buttons with D-PAD

MrCjxer, 2024-07-14:
Hi again
i don't want use another Arduino nano for max 16 buttons
and i can't edit the source code, CUZ i'm noob :)

" If you decide to go the hard way, it is possible to stack 3 8bit shift register chips (parralel in serial out) and utilize 24 buttons including a D-pad "
if i want to do that, what should i do ?

[image]

can i use this shift register module ?!

1 - i want 8 button for H Shifter ( not a analog H-shifter ) CUZ i designed my H shifter and i can't change it )
2 - i want 2 button for sequential shifter on back of wheel
3 - i want 11 or 12 button on wheel ( like logitech or ... )
4 - if it possible have a D-PAD in addition to the 3 things I mentioned above

at the end 21 button (8 for H shifter - 2 for sequential shifter - 11 for on wheel ) + 4 D-PAD

the guy you did mention :
https://github.com/johanneslerch/Arduino-FFB-wheel

but i didn't understand what he did

ranenbg, 2024-07-14:
You seem to be missing to understand many things. You need to do some research about shift register chips and how they operate. The one you suggested has an i2C interface and you can't use that one. 

Data, clock and pull are signals required for the operation of the shift register.

MrCjxer, 2024-07-14:
yeah, i told you i'm noob :)

so tell me for 3 things I mentioned, what i need ? ( software or hardware )
i want work on it hard and search for all i need

and with 6 :DT_sh ---- 7: CLK_sh ---- 8: PL_sh how many button i can use?
thank you

> 1 - i want 8 button for H Shifter ( not a analog H-shifter ) CUZ i designed my H shifter and i can't change it )
> 2 - i want 2 button for sequential shifter on back of wheel
> 3 - i want 11 or 12 button on wheel ( like logitech or ... )
> 4 - if it possible have a D-PAD in addition to the 3 things I mentioned above
> 
> at the end 21 button (8 for H shifter - 2 for sequential shifter - 11 for on wheel ) + 4 D-PAD
> " If you decide to go the hard way, it is possible to stack 3 8bit shift register chips (parralel in serial out) and utilize 24 buttons including a D-pad "

MrCjxer, 2024-07-19:
Hi again
bro any advice ?
for 21 or 22 button ?

how many shift register i can use on default FW ? 4 shift register or more or less ?

ranenbg, 2024-07-19:
Without any modification of FW, you can stack 2 8bit shift registers to get 16 buttons, or use arduino nano. With small FW mod you could utilize all 28 buttons that are available by HID, that is 3.5 8bit shift registers. By significantly moding FW and also its HID descriptor part, you can go up to 64 buttons.

MrCjxer, 2024-07-20:
hi again
thank you for your guidance <3

1 - if I want use XY Shifter, can I use it with 2 shift register? (8 button with XY shifter and 16 buttons with 2 8bit shift registers)
or just supported one function?

2 - or if I want use XY shifter, can I use it with 4x4 button matrix? (8 button with XY shifter and 16 buttons with 4x4 button matrix)

=====================================================
>With small FW mod you could utilize all 28 buttons that are available by HID, that is 3.5 8bit shift registers
> By significantly modding FW and also its HID descriptor part, you can go up to 64 buttons, that is 8 8bit shift registers.

but for small FW mod or significantly modding, need I change something on FFB-gui?

https://github.com/ranenbg/Arduino-FFB-gui

MrCjxer, 2024-07-23:
hi again
sorry bro to bother you again
can you tell me about three questions i asked?
I want to buy the components

>1 - if I want use XY Shifter, can I use it with 2 shift register? (8 button with XY shifter and 16 buttons with 2 8bit shift registers)
or just supported one function?
>2 - or if I want use XY shifter, can I use it with 4x4 button matrix? (8 button with XY shifter and 16 buttons with 4x4 button matrix)
>3 - If I go with " Button box - for Arduino Nano/Uno ", how many buttons will remain directly to Arduino Leonardo?
>

@johanneslerch
bro sorry to bother you too
can you help me with three questions?

and this 
>but for small FW mod or significantly modding, need I change something on FFB-gui?
>https://github.com/ranenbg/Arduino-FFB-gui
>

thank you

johanneslerch, 2024-07-23:
Check the wiring diagrams. They show you what is used for what purpose. XY shifter is attached to analog buttons A4, A5. You can also see that if not used as shifter, they are mapped to buttons. How many buttons there can be, is just counting them in the wiring diagram. I leave that as exercise for yourself.

Note that XY shifter must be attached to an analog input. You can't connect them to shift registers.

MrCjxer, 2024-08-03:
> Check the wiring diagrams. They show you what is used for what purpose. XY shifter is attached to analog buttons A4, A5. You can also see that if not used as shifter, they are mapped to buttons. How [...]

Hi again and thank you bro
But the wiring diagrams very confusing for me 🥲

Yeah i know for XY shifter i should use analogs but i mean when i use XY shifter, can i use two shift register beside XY shifter?

And i bought the Arduino NANO for buttons box
Can you tell me, when i use it for buttons box, how many buttons can i have on the Arduino Leonardo as directly?🤔

I mean when i use 230cx.hex for Leonardo and connect Arduino NANO to the Arduino Leonardo, how many buttons can i have on the Arduino Leonardo as directly?
Even i want use A3,A4,A5 for normal buttons if there is any chance
 
Thank you 🙏🏻😊

ranenbg, 2024-09-09:
If you use Arduino nano as button box, with v231 HEX you can have 3 extra buttons on A3, A4 and D12.

closed 2024-09-09 by ranenbg