ranenbg/Arduino-FFB-wheel #58: Arduino Nano

MrCjxer, 2024-07-31:
Hi
I want to use an Arduino Nano for button box, but which version I should use?

1 - Arduino Nano CH340
2 - Arduino Nano V3.0
3 - Arduino Nano V3.1
4 - LGT8F328P Nano

or all are the same?
or maybe can I use 1 and 2 and 3 except 4?

thank you

ranenbg, 2024-07-31:
Hi, any nano with ATmega328 will work. They usually come with a Chinese USB-to-serial converter chip CH340, because the MCU does not support it natively.

MrCjxer, 2024-08-03:
Hi
Thank you

I have a new problem bro 😅🥲
I used FW 230cx.HEX for Leonardo
But when i uploaded it to board and open the GUI, after pass the steps, it goes directly to the main GUI app !

I mean when i choseed the Leonardo at this step :

[image]

It goes to this page :

[image]

I want config the AXIS and buttons or others like this :

[image]

What should i do ? 🤔

ranenbg, 2024-08-03:
Hi, you're welcome. Please, don't abuse this issue with another question, if I already answered what you needed in the first place. You close this one as solved and then make a new issue with a new question. I will answer your new question this time and close the issue. If you still have questions, please make a new issue.

This is not a problem what you're showing, this is how I designed my firmware/software. It's supposed to work like that, as you only need to configure the pink window once correctly on a single PC and that's it. This is just the link that makes sure you're using my firmware with my GUI. On the left side there are inputs that my GUI is expecting to see and in the right side is the actual HID device (usualy arduino with some of my HEX), but in general it may be any other joystick that exists. In principle, you can link any GUI axis with any HID axis, or any GUI button with any HID button, but that's simply irrelevant, as in your games you can map any axis or button to any function that you would like. Only for the sake of simplicity, I've put an example of how a pink window should be linked, to avoid confusion, as you can map axis to a button for example , or button to hat switch and so on..It is possible to reset the config such that you do the pink window again, but I'm not gonna explain that here at this moment.

closed 2024-08-03 by ranenbg