ranenbg/Arduino-FFB-wheel #59: 3-phase BLDC Motor

I2m0roI, 2024-08-01:
Hi, 
has anyone tried to get a 3-phase motor to work? 

I'm currently planning on using one from an old hoverboard, but can't find any motor controllers with direction-specific PWM inputs. 
They all use one PWM / 0-5V input for speed and another for DIR (presumably pulled high internally, switch to GND to change direction).

Could I just connect both PWM outputs of the arduino to the corresponding input of the controller and use the 'PWM direction pin' for switching direction?

Best regards

I2m0roI, 2024-08-01:
Well, what a few hours of sleep can do for you... Found the PWM settings, solved.

closed 2024-08-01 by I2m0roI

Notriatv, 2024-08-17:
so.. did you got it working??

I2m0roI, 2024-08-17:
Yes I did. Just needed to switch to 'PWM+Dir' in the software.
But the motor controller (Walfront 5-36V from A***on) broke when I tested the torque with a 24V/5A power supply.
Still looking for a better replacement.

Notriatv, 2024-08-21:
can you provide more info about that? like some pics with wirings schemes?