ranenbg/Arduino-FFB-wheel #60: reset the config and calibrations

MrCjxer, 2024-08-05:
Hi
I want to reset the config
I mean I want to see the pink window for config and change buttons again
What should I do?🤔

[image]

*Note : i didn't see this pink window never 🤔 it's just automatic did the job 😅 but I want to change it and reset it
I am using 230cx.hex of your FW

Marcos2BR, 2024-08-07:
in dir. ...._\ranenbg-Arduino-FFB-wheel\wheel_control_v2_6_0\data_ 
delete file "Arduino Leonardo wheel v4" or "Arduino Leonardo wheel v5" 

This is the file where the button map is located.

MrCjxer, 2024-08-07:
> in dir. ...._\wheel_control_v2_6_0\data_ delete file "Arduino Leonardo wheel v4" or "Arduino Leonardo wheel v5"
> 
> This is the file where the button map is located.

Hi and thank you
I did delete files you said but it happened :

[image]

 and i still can't map buttons :(

Marcos2BR, 2024-08-07:
just config the COM PORT in file "COM_cfg.txt" 
See more in **Troubleshooting - first time run** :  [COM_cfg.txt](https://github.com/ranenbg/Arduino-FFB-gui?tab=readme-ov-file#troubleshooting---first-time-run)

ranenbg, 2024-08-07:
Ok, I have to involve here. Plsease don't delete Arduino Leonardo wheel v5, file as this is what GUI expects to see in data folder. I'm not sure if you can create it yourself, it should be possible but I didn't check it. It must be named exactly like that, or the gui will hang in the startup like that.

Why do you want to change these mappings? Can't you just rewire your buttons such that you see it in GUI as you want? This is onky link between HID and mu GUI, you are not remaping buttons as seen by windows (HID).

Marcos2BR, 2024-08-07:
I'm sorry, I was just trying to help.
Before, I did tests on the protoboard and tried to use a potentiometer for the X axis, sometimes I needed to reset the mapping and I think I deleted the file...

ranenbg, 2024-08-07:
Yeah, no worries, I understand. It's just something I have not experimented with myself. I did originaly create the mapping using the pink window in the first place. This is not my code, it's just a HID library that I'm using. It has it's downsides, but this is how it is at the moment, it's pretty much dictated by that game controls plus lib. Without it, GUI would still work but one could not see any axis or buttons, as it reads directly the HID values from the device.

Just spend some time to rewire your buttons and you're good to go.

MrCjxer, 2024-08-17:
> Ok, I have to involve here. Plsease don't delete Arduino Leonardo wheel v5 file, as this is what GUI expects to see in data folder. I'm not sure if you can create it yourself, it should be possible but I didn't check it. If you do manage to create it by yourself, the mapping file must be named exactly like that, or the gui will hang durring the startup like that.
> 
> Why do you want to change these mappings? Can't you just rewire your buttons such that you see it in GUI as you want? This is only link between HID and my GUI, you are not remaping buttons as seen by windows (HID). The games are looking at HID, so you should remap by hardware connections, it's very easy, I have documented which arduino nano pin corresponds to which HID button. What you're trying to do has no meaning, as I said before.

Hi
i want to change some buttons.
Like the handbrake axis, I don't it . and i want to remove it or change the buttons
may i have the ping page or not?

i am use your HEX for my Arduino Lenonardo
thank you

MrCjxer, 2024-08-17:
> I'm sorry, I was just trying to help. Before, I did tests on the protoboard and tried to use a potentiometer for the X axis, sometimes I needed to reset the mapping and I think I deleted this file...

hi
but it did not work for me :(
still need to get the ping window

MrCjxer, 2024-08-21:
@ranenbg

hi please HELP me :(
I need to remap the PINs and buttons
I was waiting for weeks :(

just tell me how to go in pink window and remap all PINs
thank you

ranenbg, 2024-09-09:
Either get a new arduino or just connect A4 to gnd.

closed 2024-09-09 by ranenbg

MrCjxer, 2024-09-10:
BRO please! :|

can you just tell me how I should reset the configs and see the pink Window?!
it is not about just the A4 pin I want to change some button and ANALOG pins
all wires connected with shield, and I can't change it Hardware wise.
please just tell me how should I go in pink window, if it is impossible just tell me << it is not possible >>

THANK YOU ...