ranenbg/Arduino-FFB-wheel #61: enhancement request : MOTOR_ENABLE_PIN with PWM+-

Marcos2BR, 2024-08-07:
Hello Milos,
First of all, thank you very much for your excellent work.
I'm using your firmware on a Vintage AVB Wheel Force Feedback GC-FBW1

I don't know much about C, C++, just the basics to turn comments on/off and understand the basics of the code.
Following your tips I was able to compile HEX with arduino-ide by mounting the library in the "portable" folder.
In config.h I enabled
_#define USE_QUADRATURE_ENCODER ,  USE_HATSWITCH , USE_BTNMATRIX  , AVG_INPUTS  ,  USE_AUTOCALIB,  USE_CENTERBTN , USE_EXTRABTN_ and everything works fine. 
Long live my dinosaur steering wheel and its 8 buttons + hatswitch + accelerator + brake + forcefeedback

**BUT** I feel the steering wheel a little hard at rest,
I saw that you gave the tip: "_if your motor is very hard to turn_..." (Motor driver wiring) 

So I thought I'd ask you to implement "MOTOR_ENABLE_PIN with PWM+-" , something like : 
  _//vsulako code
#ifdef MOTOR_ENABLE_PIN
  if (force!=0) //or other value
    digitalWriteFast(MOTOR_ENABLE_PIN, 1)
  else
    digitalWriteFast(MOTOR_ENABLE_PIN, 0);
  #endif_

I believe this will make the steering wheel turn smoother with low pwm, similar to Emc, Visulaco, RFR, etc

Sorry for the excessive blah blah blah, honestly, I'm thrilled to see my old school steering wheel coming back to life, and I'm also happy to let the local community know that there are alternative firmwares besides RFR and EMC.

ranenbg, 2024-08-07:
Hi dude, you're welcome.

Have you tried the the "tip" as the workaraound first? I'd like hear how
that went for you.

I'll think about how to implement your suggestion, I could utilize the
unused pin11 which is otherwise a direction pin, when pwm+- mode is used.
In principle, it's a very easy fix in the firmware, there should be plenty
of memory left for it.

Cheers,
Milos.

Marcos2BR, 2024-08-12:
> Hi dude, you're welcome. Have you tried the the "tip" as the workaraound first? I'd like hear how that went for you. I'll think about how to implement your suggestion, I could utilize the unused [...]

Hi Milos, the mechanics of my steering wheel have a 40:1 reduction , Maybe that's why at rest it feels "heavy" (with the engine braked)....
I made a connection with diodes in ENABLE (attached photo) and it was very smooth at rest and small turns.
I did not measure the PWM R/L time with an oscilloscope, but I imagine that below 20% the BTS output is still in tri-state.

[image]

closed 2024-09-09 by ranenbg