ranenbg/Arduino-FFB-wheel #64: AS5600 Magnetic Encoder

joeyda, 2024-09-02:
Hi, I have a brushless motor with an AS5600 magnetic encoder and an arduino pro micro, is it possible to work with your firmware using i2c?

ranenbg, 2024-09-02:
Hi, at the moment AS5600 is not supported in my firmware. I have no plans to implement its support in my firmware, as this type of encoders is not suitable for the ffb wheel due to not good enough response, speed, resolution and noise level. It is used mostly for digitizing pots, knobs or basic angle readout.

joeyda, 2024-09-02:
Good to know, thank you!

joeyda, 2024-09-04:
Thank you for the additional info. I was looking into making a small handheld wheel with a brushless gimbal motor since they are so quiet and tiny. I have a motor driver and was going to use simple FOC to control it

ranenbg, 2024-09-09:
Yes, I see why you want AS5600 in the controller like this it is more convenient due to the small space, but I still think an optical encoder is far superior. It will have higher resolution, no noise at all, and instant reaction speed plus you can easily add some gears to it, add a DC motor and a small h-bridge like LN298 and you're good to go.

jairov96, 2024-09-10:
That's dope, thank you very much!

Anywhere I can donate/buy a coffee to the cause?

ranenbg, 2024-09-10:
You're welcome. I still have to do some testing, but if you wish to donate there is my paypal in the GUI near bottom part of window.

ranenbg, 2024-11-06:
I'm closing this one, as I have already released firmware v240w firmware which supports AS5600 magnetic encoder via i2C bus.

closed 2024-11-06 by ranenbg