ranenbg/Arduino-FFB-wheel #65: Adding trigger buttons

HOMECODERNEO, 2024-09-06:
Building and testing the device I had an idea to add to the firmware and GUI program the possibility to configure some buttons as triggers (provided that in their place in the “matrix” will stand toggle switches place buttons). That at the events on-off and off-on events there was a short pulse of 200-300ms simulating the button presses, while the toggle switch will simply change its position without remaining clamped by the button as it is now. I don't know if I managed to describe the essence of the idea (I apologize as I translate through a translator) and how this idea sounds adequate in relation to your project.

ranenbg, 2024-09-06:
Hi, why not just use a normal togle switch and connect it to two buttons?

HOMECODERNEO, 2024-09-07:
For example, in the same ETS2 if I use the turn signal switch from the car in my homemade system, it turns out that to turn on and off the turn signals I have to move the lever twice, it does not look realistic. After all, this switch has a fixed position, not pressed. This is a basic example, and there are several other situations like this.

HOMECODERNEO, 2024-09-07:
This is a pretty specific case, and worth considering as an idea for improvement if it has the right to exist in the project.

ranenbg, 2024-09-07:
It's a good idea, I agree. On the coding side of things it's not simple at all to implement it, as I would need to use timers and quite some memory space for it. Also, I don't want to implement a feature in firmware that will be usable only in ets2.

I read your first post a few times and I'm not sure that I understand exactly how you want these special buttons to operate. I'd appreciate if you explain again in more details, with an example of function in the game that you want to simulate and how the firmware should handle that. How does ets2 handle sterring indicator lights?

To me it seems that what you want can be easily done by using correct hardware. There are many types of switches, like the ones with 2 poles, 3 positions that always go back to center position.

ranenbg, 2024-09-09:
It may be possible to such a feature implement on some buttons that are directly on arduino, but definitely not in the button matrix. It uses a special algorithm to scan all rows and columns to find which buttons are pressed, there can't be any timers there.