ranenbg/Arduino-FFB-wheel #66: Problem while calibrating

EbrSiami, 2024-09-10:
Hello sir, first of all I have to thank you for your very nice and practical project because it solved one of my biggest problems. Actually, I did everything right and now everything works great, but my big problem is that unlike the pictures, no screen like the one for calibrating the controls (Wheel_control_gamecontrols_setup.png) opens for me. I wanted to ask how can I solve the problem? Thank you very much
[image]

ranenbg, 2024-09-10:
Hi, you are very welcome.

I'm not sure that I understand what the problem is exactly. You said yourself that everything is working well? Can you please elaborate on what you have done so far and what kind of hardware are you using?

I'm not able to open the png you attached, but I suppose you are talking about game controls plus library, which has a pink setup window. It has nothing to do with calibrating any axis. You can calibrate the analog axis directly from GUI.

EbrSiami, 2024-09-11:
Yes, exactly. Im using an Arduino Leonardo and as I said Everything connected is great, except for a few things. Actually, I want to change the buttons and other things so that they work properly as well. I tried to try other ways like changing some codes etc. but I didn't succeed.I want that pink page to open for me so that I can do this, but unfortunately this page does not open for me. Even on the first run.

ranenbg, 2024-09-11:
Ok dude, I knew who you are from the beginning, but I do appreciate wanting to do a fresh start.

Mapping the pins and axis to buttons is done through code and not through pink window. I worked extremely hard for many years to get it to this stage and all buttons and axis are mapped in that way as a result of very hard thinking. Every single pin on all 3 arduinos is used for something. Furthermore, those mappings are not static, some buttons are dynamically changing for each fimware option/version. It took me a lot of time to figure out how to do this, such that end user only needs to comment or uncomment some lines in config.h and that's all.

If you wish to change mappings, there are several layers that have to be take care of. This is a very difficuilt task for a beginner programmer. On the other hand, simply rewiring the button to a different pin, which is already assigned in code to be a button, or axis is much much easier.

I would appreciate if you show me your hardware, you mentioned that you are using some shield for axis and buttons? Just remove it and make the new connections as you wish yourself.

MrCjxer, 2024-10-29:
Hi

I need that too
please just tell us how to change mapping or change the buttons

closed 2024-11-06 by ranenbg