ranenbg/Arduino-FFB-wheel #68: flight firm request.. also. What a shame!!

Notriatv, 2024-11-01:
Hello Rane, I have a question.. would it be possible to develop a version with an additional PWM output? It might be necessary to remove some features to free up memory for this, but the idea would be to create a specific version for flight controllers with support for feedback on two axes. I managed to feel FFB effects like side wind in MSFS20 through the steering wheel using a plugin called 'XPForce' (only 10min trial), but there is only one active PWM output in the firmware you provide. What do you think? Would it be possible to create a firmware like this? Thank you in advance.
 
also it's a shame that such a great piece of work receives so little recognition. : )
If you don't mind, I made a video showcasing your work and posted it on YouTube so that people can benefit from your work, which I believe has great potential. Here is the link if you want to check it out. https://youtu.be/eAml14yTIjk

ranenbg, 2024-11-01:
Hi man, tnx very much for your youtube video. I understand a little bit of spanish, so I got the gist of what you were saying in portugese. You covered most important stuff very well, and it was a cool to watch it.

Although I have two ffb axis in HID, It's not trivial to add a 2nd ffb axis via pwm output. I was busting my head before with this. This suggestion is a legit expansion of firmware and it's definately possible. The execution however, on arduino in my oppinion would not have satisfactory results. I'm talking about flight stick axis with pots and low frequency 8bit PWM (very noisy and low resolution FFB). Maybe I could use two AS5600, but still PWM would be the major limitation, as there are not 4 PWM pins with 16bit resolution. Arduino only has one 16bit timer, associated with pins D9 and D10. Another option is to make 2 channels with 16bit PWM+DIR signals, but then bts7960 can't be used. One more thing comes to mind - to use external DAC's for analog FFB output. That would allow usage of AASD and servo motors, which is a huge increase in cost. Or to have 16bit FFB on X-axis and 8bit FFB on Y-axis, with PWM+- (h-bridge) signals..There are several options how to do it, but none of them would retain the quality of the firmware as it is now, so I'm still undecided.

OpenFFB firmware for stm32 already has support for more than 1 ffb axis and various motor drivers like dual odrive and so on..It is suited much better for the flight stick and it's worth looking into.

Notriatv, 2024-11-03:
Hi bro, I'm glad that you liked it! I'll be making many more detailed videos about it. It’s really cool, and I hope more people discover your project.

So yeah, the idea is to do it at the lowest cost possible and in the easiest way as well. Using servos will make things a little more complex to deal with. I don’t know how effective an 8-bit FFB on the Y-axis would be (maybe it’s enough for a flight simulator). Anyway, I’ll soon create a mechanical part for it and will use a Bourns multiturn pot, or I might test an AS5600 in pot mode using a spring for the Y-axis and FFB for the X-axis. It should definitely be cooler than the Logitech ones. Let’s see how it goes!

Also, there is a guy (Gagagu Arduino_FFB_Yoke) working on a 2-axis FFB support for the ATmega32U4, and he’s also doing a great job . I think people underestimate the power of this chip. i know tha any stm32 is more advanced and better than it but I believe it’s enough for many users—just my opinion, you know, lol! 

big tnx!!!

ranenbg, 2024-11-03:
I'm pretty sure he makes 8bit ffb on both axis with 8bit PWM, if he uses h-bridge 2ch config for each motor. This is not what I want to do, so I need to think more about how to do it properly.

Avoid using pots for any ffb axis, AS5600 is a very good alternative, if you can't use optical quadrature encoder.

Notriatv, 2024-11-04:
sure, everything you create I'm ready to test. thanks

closed 2024-11-04 by Notriatv