ranenbg/Arduino-FFB-wheel #70: Question about shift register

hikineet98, 2024-11-09:
Hello ranenbg. Thank you for this great project.
I would like to ask about shift register.

Setup: 
-Arduino Leonardo
-bts7960
-Arduino nano v3.0 ch340 (shift register)
-Firmware: 241chfx

Behaviour:
Button press not work on Tx, d3 pin and other pins at Leonardo side (eg: button to switch to reverse gear at pin 4) when firmware 241chfx used. But those pins do work when i switched to firmware 241ahx.

I don't really know how the shift register work, but when i measured the voltage on each pin, pins that work show ~5v when button not press and 0v when pressed. Pins that not work is opposite. 

One more thing is about Arduino nano firmware upload. I saw an old issue when you suggested upload using atmega328p (old bootloader). But when i upload using old bootloader, it just freezed at uploading state. Upload using atmega328p works immediately.

About those buttons that work. When i pressed 4 or more buttons, all the buttons in the nano just pulsing.

Also I'm using a long phone cable to connect from Leonardo to nano, and a breadboard  to connect all gnd and vc pin. Could it be too long connection or maybe some thing not properly connect cause this problem?

Please enlighten me.

ranenbg, 2024-11-09:
Hi, you're welcome.

Tnx for reporting this issue. I have not seen it before and I can't tell right away why this is happening. I will have a look at this particulair HEX version on Leonardo and let you know. My initial guess is that it has something to do with nano v3.0, as I didn't use those new ones, only older ones that come with "old bootloader".

The "shift register" code for nano has internal pullups enabled which means that all pins will be high (5V) - not pressed, until shorted to gnd (0V) - pressed. It is wierd if that's not the case on nano.

Pin Tx is not for button, it is one of encoder pins. Pins D3, D4 are buttons but not in all HEX. I will look into this, to see if I can reproduce the issue and see what can be done about it.

hikineet98, 2024-11-09:
Thanks for the quick reply.
My wheel is totally usable and i have already enjoyed some games so please take your time.
But you know, the more buttons the better :))

ranenbg, 2024-11-13:
Ok, I have managed to reproduce the issue where on HEX v241chfx, pins on nano 3.0 (with buttonbox code) Tx, D3 and D4 are not working. I'll see if anything can be done about that, if so, I'll update the buttonbox code for that nano version. I had to set ATmega328P in order to get it to upload, as the "old bootloader" did not manage to upload to this 3.0 board. It has a new bootloader and there is something different, I don't know what.

Also, you are correct, pin D4 on Leonardo for the reverse gear button is not working. I have to fix that somehow in the following fw version, tnx once again for reporting it.

hikineet98, 2024-11-14:
Thanks for the quick update! I'll test it when i get home and close the issue if there's no other problems.

closed 2024-11-14 by hikineet98