ranenbg/Arduino-FFB-wheel #73: Feature request Ability to add 16 bit ADC

moezzasif, 2024-11-22:
Please add the ability to use 16 bit adc for FFB software

ranenbg, 2024-11-22:
Hi, first of all, if you want me to even consider implementing anything in the firmware/software, you'll have to be much more specific about your request.

The firmware already supports ADS1015 (12bit) and ADS1115 (16bit) ADC chips. To enable the 16bit one, you just need to comment 1 line not to use the 12bit one and uncomment the other one for ads1115. But it's not so simple, as only X and Y axis are 16bit in HID, and others are 12bit. X-axis is already used for encoder and Y axis is reserved for load cell brake wich has 24bit HX711 chip, or can be set to pot and arduino 10bit analog input is used or ads1105 12bit. Another downside of using ads1115 is that it is very slow. It will slow down the entire loop, making all other inputs to run slow, including reducing the FFB frequency to something about 100Hz or so, depending on how many inputs you want to read from it. This is why I didn't compile a HEX with such an option. 

Just out of curiosity, which ADC did you have in mind? There are many available with various comms, i2C, SPI, i2S and so on..

ranenbg, 2024-11-22:
Yes, I can see that, however, one has to start from somewhere. You are welcome.

Porting my firmware to stm32 is a huge amount of work. It requires completely rewriting most of the code from scratch and I have no plans of doing that. Others have done that already and there is plenty of available FFB firmware for stm32F407 like: MMOS, EMC, VNM, or OpenFFBoard. The last one I strongly  recommend, as it is the most advanced of all and one can use it even to make a direct drive wheel, once paired with the special motor driver board done by the same guy.