ranenbg/Arduino-FFB-wheel #74: problem with buttons

Eyadserry, 2024-11-30:
I don't know if this is a problem. When one button is pressed and I try to press another button, the other button doesn't work. Instead, the first button toggles .

ranenbg, 2024-11-30:
Hi, what firmware are you using? Which buttons? Are you talking about buttons on arduino, or nano or button matrix?

Eyadserry, 2024-11-30:
v240chtx, I'm using the button matrix on leonardo.

ranenbg, 2024-11-30:
Yeah, not sure, doesn't sound like normal operation that I intended, will need to investigate. Tnx for eporting it.

Eyadserry, 2024-11-30:
https://github.com/user-attachments/assets/fd9dda26-ab4d-487e-8a85-6571e560e8b5

As you can see in the video, I pressed button number 8. While it was pressed, I also pressed button number 0, and then button 8 started toggling. When I released my finger from button 8, button 0 lit up. I did the same thing again."

Eyadserry, 2024-12-01:
if you have other questions to solve the problem, I would be pleased to answer.

ranenbg, 2024-12-01:
Tnx, for now this is enough info.

Eyadserry, 2024-12-01:
ok, I'm gonna close the issue.

closed 2024-12-01 by Eyadserry