ranenbg/Arduino-FFB-wheel #75: Magnetic encoder magntek-MT6816

mac-gar, 2024-12-13:
Hello.
I would like to ask for some advice from you or the community gathered here. How to use the MT6816 magnetic encoder in this project. I see that the code has support for AS5600, which connects via I²C and my encoder has several interfaces: ABZ, UVW, PWM or SPI

I would like to use its parameters to the maximum, because they are very good:
https://xonstorage.blob.core.windows.net/pdf/magntek_mt6816ctstdr_apr22_xonlink.pdf

14-bit Core Resolution
Output Propagation Delay <2 us
Maximum Rotation Speed ​​25,000 RPM

If it is possible to add such support in this project, I would be very grateful. If for various reasons it is not possible. Please give me some advice, can I send these parameters to Arduino micro pro in another way.

ranenbg, 2024-12-13:
Hi, this seems like a very good  anglular sensor. Tnx for suggesting it.

After reading quickly through its data sheet that you provided I can see that it has an ABZ output of up to 1024PPR. In principle you can already use it like that, once you configure it in ABZ mode. This means you have 4096CPR which is good.

On the other hand, to utilize its full 14bit resolution by implementing SPI is possible. But if you want me to add I'd appreciate very much an example code for arduino that reads angle over SPI with multiturn capability. Then I could chech if I can fit that code into firmware.

mac-gar, 2024-12-16:
I don't know how to use the code from this source, but there is a driver for this encoder with an example

https://github.com/simplefoc/Arduino-FOC-drivers/tree/master/src/encoders/mt6816

https://github.com/simplefoc/Arduino-FOC-drivers/tree/master/examples/encoders/mt6816/mt6816_spi

ranenbg, 2024-12-16:
That is exactly what I needed, tnx.