ranenbg/Arduino-FFB-wheel #76: driver for bldc???

Notriatv, 2024-12-13:
Hi Rane,
Regards!

I’m really satisfied with ffb that my DIY belt-driven/IBT-2 provides. But I was thinking about using Arduino with your program to control some BLDC motors, such as hoverboard motors or any other 3-phase motors. But I’m not sure which driver would work with Wheel Control for Arduino. Which pinout should I use to connect the boards? Which mode should I run? (PWM+DIR? 0-50-100? RCM mode?)

I’m really lost on how to do this and whether it’s even possible. let me know if you know a good driver to run a bldc motor

I’ve been looking into some BLDC drivers, and here’s what I found that could be a cheap option. What do you think about them?

1
[image]
  https://pt.aliexpress.com/item/1005007634026470.html?spm=a2g0o.detail.pcDetailTopMoreOtherSeller.4.7bedHfIGHfIGjY&gps-id=pcDetailTopMoreOtherSeller&scm=1007.40196.366991.0&scm_id=1007.40196.366991.0&scm-url=1007.40196.366991.0&pvid=27899184-88d6-4b12-bcdd-1cecf4220a55&_t=gps-id:pcDetailTopMoreOtherSeller,scm-url:1007.40196.366991.0,pvid:27899184-88d6-4b12-bcdd-1cecf4220a55,tpp_buckets:668%232846%238107%231934&pdp_npi=4%40dis%21BRL%2174.51%2122.35%21%21%2185.56%2125.67%21%402101efeb17341259448666382eb710%2112000041584593943%21rec%21BR%21803869506%21XZ&utparam-url=scene%3ApcDetailTopMoreOtherSeller%7Cquery_from%3A

2
[image]
https://pt.aliexpress.com/item/1005006419395042.html?spm=a2g0o.detail.pcDetailBottomMoreOtherSeller.18.22d0FYQqFYQqhK&gps-id=pcDetailBottomMoreOtherSeller&scm=1007.40196.366991.0&scm_id=1007.40196.366991.0&scm-url=1007.40196.366991.0&pvid=5b527bfd-2e90-46cf-847e-b70041d86339&_t=gps-id:pcDetailBottomMoreOtherSeller,scm-url:1007.40196.366991.0,pvid:5b527bfd-2e90-46cf-847e-b70041d86339,tpp_buckets:668%232846%238107%231934&pdp_npi=4%40dis%21BRL%2167.78%2135.24%21%21%2177.83%2140.47%21%402101efeb17341259800057061eb710%2112000037142323608%21rec%21BR%21803869506%21X&utparam-url=scene%3ApcDetailBottomMoreOtherSeller%7Cquery_from%3A

3
[image]
https://pt.aliexpress.com/item/1005007442976530.html?spm=a2g0o.cart.0.0.4daa7f0667MNlb&mp=1&gatewayAdapt=glo2bra

4
[image]
https://pt.aliexpress.com/item/1005006604285603.html?spm=a2g0o.cart.0.0.4daa7f062nJK9o&mp=1&gatewayAdapt=glo2bra

useful info i guess https://docs.simplefoc.com/

That's it. I know that ATmega32 isn't the best MCU for a DD Wheel, but one day I'd like to give it a try—just to see how bad it could get XD

thanks in advance!!

ranenbg, 2024-12-13:
Hi, bldc motor driver is extremely complex topic that goes outside of the scope of my original purpose for the ffb firmware.

Out of these 4 drivers that you listed, only 4th one seems promising since it has FOC. I'm afraid that it's not possible to get good result with cheap bldc motor drivers. Most of them only have speed control, and you need torque control for ffb wheel. If the driver has pwm0-50-100 input mode (like some howerboard motor drivers) you can use that with my firmware. Maybe some will have even an analog torque input. Another option is comertial AASD 30A 3 phase motor driver and some 10Nm mige motor, this driver has analog torque input.

My advice for you is to join OpenFFboard discord and ask around how to make ffb wheel with howerboard motor + moded O-drive + stm32f407 with OpenFFb firmware. This is the cheapest DIY DD wheel one can make. There are some techical manufacturing chalenges, but it's manageable for most hobby guys.

Notriatv, 2024-12-14:
yes.. I see.. bldc is great but also a really complex thing to work.. 
well thanks for your tips!!
 also FFbeast for ODESC is other good option for hoverboard motors..
THXX!!

closed 2024-12-14 by Notriatv