ranenbg/Arduino-FFB-wheel #77: Out Of Memory

namcktv, 2025-01-14:
#ifndef _CONFIG_H_
#define _CONFIG_H_

//------------------------------------- Options ----------------------------------------------------------

//#define USE_VNH5019				// Pololu dual 24V DC motor drive (partially inplemented)
//#define USE_SM_RS485				// Granite devices simple motion protocol (not implemented yet)
//#define USE_LCD					// milos, LCD via i2c (not implemented yet)
//#define USE_ADS1015       // milos, uncomment for 12bit pedals, commented is 10bit from arduino inputs (can not be used with AVG_INPUTS)
//#define USE_DSP56ADC16S			// 16 bits Stereo ADC (milos, can not be used with USE_SHIFT_REGISTER)
#define USE_QUADRATURE_ENCODER		// Position Quadrature encoder
#define USE_AS5600          // milos, uncomment to enable magnetic encoder via i2C instead of optical encoder
//#define USE_ZINDEX          // milos, use Z-index encoder channel (caution, can not be used with USE_ADS1015, USE_MCP4725 or USE_AS5600)
//#define USE_LOAD_CELL				// Load cell shield // milos, new library for LC
//#define USE_SHIFT_REGISTER			// 2x8-bit parallel-load shift registers G27 board steering wheel (milos, this one is modified for 16 buttons)
//#define USE_DUAL_SHIFT_REGISTER		// Dual 8-bit Parallel-load shift registers G27 board shifter  (milos, not available curently)
//#define USE_SN74ALS166N          // milos, uncomment to use 3x8bit parralel-in serial-out shift register chips for 24 buttons, otherwise it's 16 buttons with ard nano-buttonbox (must be used with USE_SHIFT_REGISTER, but not implemented yet)
#define USE_XY_SHIFTER    // milos, uncomment to use XY analog shifter (can not be used with USE_BTNMATRIX, note that for proMicro clutch and handbrake will be unavailable)
//#define USE_HATSWITCH        // milos, uncomment to use first 4 buttons for hat switch (D-pad)
//#define USE_BTNMATRIX        // milos, uncomment to use 8 pins as a 4x4 button matrix for total of 16 buttons (can not be used with load cell, shift register or XY shifter)
//#define AVG_INPUTS        // milos, uncomment to use averaging of arduino analog inputs (can not be used with USE_ADS1015)
//#define USE_AUTOCALIB        // milos, uncomment to use autocalibration for pedal axis (if left commented manual calibration is enabled)
//#define USE_CENTERBTN    // milos, ucomment to assign digital input pin D2 for hardware wheel recenter to 0deg (caution, can not be used with USE_ZINDEX, USE_ADS1015, USE_MCP4725 or USE_AS5600)
//#define USE_EXTRABTN    // milos, ucomment to configure analog inputs on pins A2 and A3 as a digital button inputs (2 extra buttons, note that clutch and handbrake will be unavailable)
//#define USE_MCP4725      // milos, 12bit DAC (0-5V), uncomment to enable output of FFB signal as 2ch DAC voltage output
//#define USE_ANALOGFFBAXIS // milos, uncomment to enable other than encoder X-axis to be tied with FFB axis
//#define USE_PROMICRO    // milos, uncomment if you are using Arduino ProMicro board (leave commented for Leonardo or Micro variants)
#define USE_EEPROM // milos, uncomment this to enable loading/saving settings from EEPROM

#define CALIBRATE_AT_INIT	0 //milos, was 1

Sketch uses 29382 bytes (102%) of program storage space. Maximum is 28672 bytes.
Global variables use 1192 bytes (46%) of dynamic memory, leaving 1368 bytes for local variables. Maximum is 2560 bytes.
Sketch too big; see https://support.arduino.cc/hc/en-us/articles/360013825179 for tips on reducing it.
Error compiling for board Arduino Leonardo.

ranenbg, 2025-01-14:
Hi, why are you posting this as an issue?

You obviously can't upload any program that uses more than 100% of available memory.

namcktv, 2025-01-14:
I want to use AS5600, XY_SHIFTER, EEPROM but recompilation is too memory

namcktv, 2025-01-14:
What are the benefits of EEPROM?

ranenbg, 2025-01-14:
I'm not sure that commenting out EEPROM will give you that extra 2% of memory. Best is to comment out XY shifter and use another arduino just for XY shifter.

namcktv, 2025-01-14:
> Do you ever wonder why I have not compiled a HEX that already has this? Well, I guess, now you know.
why is it mandatory to use QUADRATURE_ENCODER, while AS5600 is an alternative, you can update the code so that AS5600 runs independently without depending on QUADRATURE_ENCODER option?

namcktv, 2025-01-14:
why is it mandatory to use QUADRATURE_ENCODER, while AS5600 is an alternative, you can update the code so that AS5600 runs independently without depending on QUADRATURE_ENCODER option?

namcktv, 2025-01-14:
> I'm not sure that commenting out EEPROM will give you that extra 2% of memory. Best is to comment out XY shifter and use another arduino just for XY shifter. Or use optical encoder instead of a magnetic one.

Sketch uses 27510 bytes (95%) of program storage space. Maximum is 28672 bytes.
Global variables use 1190 bytes (46%) of dynamic memory, leaving 1370 bytes for local variables. Maximum is 2560 bytes.

ranenbg, 2025-01-14:
Ok, just change default settings when not using EEPROM and you're done.

ranenbg, 2025-01-14:
Yes, it's possible. It is on my very extensive to do list..

namcktv, 2025-01-14:
#ifdef USE_AS5600 // milos, added
cQuadEncoder myEnc;
AS5600L as5600(0x36); // uses default wire.h, milos added i2C address
#endif

And comment out USE_QUADRATURE_ENCODER

Sketch uses 15406 bytes (53%) of program storage space. Maximum is 28672 bytes.
Global variables use 1131 bytes (44%) of dynamic memory, leaving 1429 bytes for local variables. Maximum is 2560 bytes.

ranenbg, 2025-01-14:
You can't just comment it out, you then leave out everything :)

namcktv, 2025-01-14:
> You can't just comment it out, you then leave out everything :) axis, buttons, ffb, the code has nothing without USE_QUAD_ENCODER
> 
> I have to work on that when I have time. Or you can do it yourself..

It is a very difficult thing for me, just hope you have time for the problem I mentioned.

ranenbg, 2025-01-14:
I'll see what I can do, but I'm not promising anything.

namcktv, 2025-01-24:
[brWheel_my.zip](https://github.com/user-attachments/files/18530378/brWheel_my.zip) I compiled and uploaded successfully but when I opened the GUI it didn't recognize the AS5600 and the project is stuck until now.