ranenbg/Arduino-FFB-wheel #79: Can use Nano's analog pins for XY shifter, throttle, clutch, brake?

NamkingWIN, 2025-01-21:
Hi, Your work is great!
I have wonder: why we dont use analog pins from Nano to control XY shifter, throttle, clutch, brake even hand brake? It will release some pins on Leonardo. With that free pins on Leonardo we can use from matrix button?
Thanks for your work

ranenbg, 2025-01-21:
Hi, thank you, you're welcome.

Even though this may sound like a good idea at first, let me explain why it's not.
1. one needs 2 arduino's for basic wheel functionality
2. reliability issues, you don't want your pedal or shifter inputs gone in the middle of a race
3. speed issues, to send six 10bit analog inputs from nano to leonardo means sending 60bits via software spi with a 90kHz clock, one needs 180 clock cycles  only to send them and this is not even counting how much time it takes nano to do adc conversion, no way could main leonardo loop run at 500Hz
4. using nano as button box as it is already, to send 16 buttons (16 bits) already slows down main leonardo loop to about 300Hz which is still acceptable as not many games have more than 300Hz ffb, but I don't want any more slowdowns
5. arduino is simply not fast enough

There, I hope this clears it.

NamkingWIN, 2025-01-22:
Thanks for your explaination. Have you think about seperating the main wheel and the shifter+pedal+button box? Or switch to stm32 controller?
But I should tell you again: your work so great and I love it. I have collecting all materials to build the FFBwheel. I am newbie so any guide with more details are avaiable? Your post on github just the wiring electronics parts.
The config I use:
#define USE_QUADRATURE_ENCODER
#define USE_SHIFT_REGISTER	
#define USE_XY_SHIFTER
#define USE_HATSWITCH 
#define USE_CENTERBTN 
#define USE_ANALOGFFBAXIS
#define USE_EEPROM
#define CALIBRATE_AT_INIT	0

The built file has 99%
Is that OK?

Thank for your time.

ranenbg, 2025-01-22:
Hi, there is no benefit in separating the firmware into wheel and pedals, as then I would have to rework the gui and maintain two firmware instances.

I have no plans to port firmware to stm32, as it requires a complete code rewrite. I don't want to re-invent the wheel. There are already works of others like mmos, emc, vnm and openffb firmware available for stm32f407.

There are plenty of examples for a DC motor + belt drive ffb wheel builds on youtube. One great builder I worked with is Orban Kokbudak, look him up. They also have a discord channel where you can find all mechanical details.

As long as you can compile and fit the code in arduino memory, it will run fine. So far, I have not seen any issues caused by low remaining dynamical memory (ram). For your options combination, just make sure to set the firmware version to 241 (converted to hexadecimal form) in order for gui to properly recognize all the features, this is also in config.h

NamkingWIN, 2025-01-22:
Config for version is below:
#define VERSION		0xF1

Is that correct?

ranenbg, 2025-01-22:
if 15×16+1=241 then yes