ranenbg/Arduino-FFB-wheel #82: Hex file  for ( Adc + xy shifter + button matrix )

Pawan419, 2025-02-05:
Hey. I wanna use adc and xy shifter at same time. But cant find this combo in any hex file.   And  Can someone tell me using adc can slow the main Leonardo board??  And button configuration is confusing me..  if I don't use nano for buttons or button matrix...  What's pinout for buttons then?  I can't find any combo with ( Adc + xy shifter + button matrix ) s + f + t . Can someone help me?

ranenbg, 2025-02-05:
Hi, I usually compile possible HEX combinations, the ones that can fit in leonardo's small 32k memory.

There is a HEX with f and t, but fitting also s can't be done as it goes over 32k limit. Yes, "s" option with ADS1015 considerably lowers the Arduino main loop. If you really want it, you can use another leonardo just for pedals that have the "s" option.

If you don't use button matrix or external button box (nano) then available buttons are listed in the firmware pinout, it's the first image in the github readme.

Pawan419, 2025-02-05:
Thanks sir.  I will find some way coz assetto corsa didn't let me use multiple hid devies for gameplay.  But I will make hall sensor pedals separately n try to make it work.
Can u tell me which version of hex file gives best ffb... I will adjust buttons n pedals with other Arduinos to keep ffb at its fastest response or am I wrong to do it ?
I really appreciate ur work 😊

ranenbg, 2025-02-05:
You are welcome. 

All firmware versions have the same ffb, however v240 versions have full 500Hz ffb, because they are most basic. You should note that final quality of ffb depends on pwm settings, so as on game ffb update rate - 60Hz iRacing, 100Hz lfs, 180-300Hz acc and so on.

My gui can only connect to 1 leonardo at a time, so you would have keep changing COM_cfg depending on shich one you want to configure. It's not a bad idea, considering you will once calibrate pedals and not have to touch it anymore. But I recommend using v240ft and having everything in one arduino.

For pedals, you csn use hall sensors, but considering noise, if done correctly sith shielding even pots are fine.

Pawan419, 2025-02-06:
I was thinking alot last night and now woke up lil confused...  Idk much abt coding.  If I m using 6 axis with Leonardo s internal ADC... Pedals and xy shifter.    Will this make main loop slow compared to external ADC ?  I wanna use 6 as5600r for 3 pedals. 1 handbrake and 2 xy shifter. But I can replace one with hx711 load cell....    And 600 ppr incremental encoder for steering.  
I have tried ai wave wheel too. It was buggy.  Affbwheel dint work well with shift registers coz ets2 wasn't reading buttons for gears well.
My main concern is.... If I use nano and configure it to give output like external ADC...  It will process data frm 6 As5600 or (5 as5600 + load cell hx711) reducing load from main Leonardo. Will this work?  N can u share ur email. I wanna ask something else too.

ranenbg, 2025-02-06:
Arduino internal ADC is quite fast, it takes about 0.1ms for a single readout, so reading 6 analog axis is still quite fast and there is even some time left even after reading encoder pulses and processing other functions.  Main loop takes anywhere between 1ms to 2ms, so full 500Hz is available.

You can use multiple AS5600 but only in analog output mode, this will be the same as if you connected 6 pots. I recommend in that case to use at least 1 of them for steering via i2C bus, that is with firmware "w" option in order to utilize its full 12bit resolution. Arduino adc is only 10bit which is fine for pedals.

Using nano to read and send analog axis via my shift register protocol has been asked many times before. I allways say that it will do no good, because sending so much data with my "software made" protocol as it is now will only slow down things way too much. It was designed to send only a few (16) bits for buttons. If one wants to send 60bits or more, i2C or SPI communication between nano and leonardo has to be implemented. This is possible, but it's a lot of work and I don't have plans to do it anyti.e soon.

You can see my email in GUI.

closed 2026-09-10 by Pawan419