ranenbg/Arduino-FFB-wheel #83: Motor recommendation.

SurArc67, 2025-02-07:
Hi, so i really wanted to build this project, but i have no idea about what motor should i choose. So can you maybe give some recommendations, and maybe add them to the docs?

ranenbg, 2025-02-07:
Hi, I can recommend Johnson RS775 dc motor. It's 300W, cheap and has enough torque, about 0.8Nm peak. You for sure need some gear or pulley ratio of atleast 1:5 up to 1:10. There are a few versions of it, try to get one with the lowest rpm if you can, as you need low rpm high torque for ffb wheel.

Good options is also RS875. Take care about voltage, at 12V these beasts are hungry for curent, so you need a psu of at least 20A. Note that 12V will draw more curent amd 24V version will draw less. They have the same power (P), which is product of voltage×curent, so you can change the ratios between the two, depending on which psu you can get.

SurArc67, 2025-02-07:
Thanks, great work btw!

ranenbg, 2025-02-07:
Thank you, you're welcome.

Some other recommendations for pulleys. You can use HTD 3M timing belt, 15mm witdh and a smallest pulley you can find for the motor, it needs to have a 4mm bore as this is the motor shaft thickness. Guys usualy design and 3D print their own big pulley for the motor shaft. It should be 8mm minimum, aluminium or cheaper version steel rod.

ranenbg, 2025-02-07:
That depends on the pulley ratio you are going to use. After the reduction from pulleys, you should have at least 250rpm max wheel rpm in order to ensure good FFB response and reproduction of higher vibrational frequencies above 50Hz.

For example, when using a pulley ratio of 1:10, your motor has to have at least 2500rpm. Usually for RS75 motors this is around 17k rpm which is not ideal, because it means it will have low torque. A good motor for an FFB wheel is the one with a very low rpm and huge torque, like the ones in the direct drive wheels.

SurArc67, 2025-02-07:
This may be a domb questions, but there are some 795 3000 RPM motors, are they suitable for this?

ranenbg, 2025-02-07:
It's not a dumb question. Yes, that is a good motor.

Notriatv, 2025-02-07:
> Hello, there are a lot of options for rpms, and i wanted to ask, what is like the optimal number of rpms?

hey guys wasup, I recently made videos showing the power of a low speed 775 (6000rpm at 24v) 0.58nm as in seller specs..)
I did a test with a 1:12 htd3m standards and a test with 20:1 gt2 patern... I find it's a surprising motor with that size

here's a link if you want to checkout. I'm speaking portoguese but you can use captions if needed

https://youtu.be/RFx5mbu9JVg?si=bEo6Odjk0ECvDuMI  htd3m

https://youtu.be/vtaQUBeWG40?si=s_NrH49Rm9fDSlK3  gt2

Notriatv, 2025-02-07:
[image]

1:12 test

but considering thermal efficiency I'd recommend a 895

ranenbg, 2025-02-07:
Tnx man :)

shubham0x13, 2025-03-18:
> Hi, I can recommend Johnson RS775 dc motor.

Will I need two BTS7960B drivers if I use dual RS775 motors?

ranenbg, 2025-03-18:
Most probably yes.

shubham0x13, 2025-03-18:
Got it, thanks! I'll plan for two BTS7960B drivers then.