ranenbg/Arduino-FFB-wheel #84: XY Shifter with Joystick module

NinjaFrozr, 2025-02-07:
I want to make an XY shifter using a HW504 Joystick module, I've wired the module's X axis to A4 and Y axis to A5. I have grounded my unused Analog pins (A2 and A3) yet i still can't get the shifter to calibrate properly. There are areas on the calibration screen that my shifter just doesn't go to (bottom right for example). I did of course move the boundaries around but i still can't really get it to go into the 8th position.  Is it not suitable to use a joystick module for XY shifter ? Or have i made some other mistake somewhere ?

ranenbg, 2025-02-07:
You didn't make any mistake, the problem is most probably with the pots themselves in the joystick module. They simply don't have enough angle range to cover full resistance swing from 0 to what ever their max resistance is.

If you have a multimeter, set it in resistance mode and measure between middle pin and one of the outer pins, move the joystick and observe how much resistance can swing. Then measure between outer pins, this is pots resistance. Middle pin should be able to reach both outer pins, if it doesn't that explains your issue.

NinjaFrozr, 2025-02-08:
Thanks for the quick and detailed response. I should probably at least find and test a spare joystick module before i consider trying your alternative solution. 

I will update tomorrow and add screenshots if can't get it to work.

ranenbg, 2025-02-08:
If you don't have a multimeter, you can use arduino. Simply make a new sketch where you read 2 analog inputs and send them to serial port using Serial.print() function. It should be something like this:

void setup() { 
Serial.begin(9600);
pinMode(A0, INPUT);
pinMode(A1, INPUT);
}

void loop() {
Serial.print(analogRead(A0));
Serial.print(" ");
Serial.println(analogRead(A1));
}

Connect your joystick middle pins to A0, A1, use gnd and 5V. Open serial monitor in arduino IDE and observe the 2 pot values.

NinjaFrozr, 2025-02-15:
I switched to using 2 potentiometers instead of a joystick module. Now i was able to set the boundaries so that my shifter can go into every position. But now i wonder if there is a way to calibrate the shifter potentiometers exactly how we can calibrate the pedals ? Because physically my x and y pots move very little and so all the boundaries have to be very close to each other as it is. It works, but it would be better if we could set the min and max positions for each shifter axis so that we can get a more sensitive xy position.

ranenbg, 2025-02-15:
Arduino analog inputs are 10bits. That meas total of 1024 steps between 0V and 5V, or if internal adc reference is enabled then from 0V to 2.56V. If you feed your pot with 3.3V or 5V this is the only way to increase the sensitivity of analog input when using lower than 5V ref for adc.

But, you're focusing your attention in the wrong direction. You can't fix everything through software. The software is precise enough. What you should do instead is build a joystick mechanics such that each pot is rotated more, for example by using some gears.

NinjaFrozr, 2025-02-15:
This is not what i meant. Perhaps sensitivity is the wrong word for it. I don't know how it works in the code but in the gui we can "teach" the software the min and max position of the pedal axis. I was asking if the same was possible for the shifter XY axis. And if not, it would be super useful to  have it. EMC has this feature and the XY shifter design I'm using is one of the most popular on Youtube. I've already gotten the shifter to work, i was just making a suggestion as the calibration is at the limit (the sliders can't move any closer together). 

[image]

ranenbg, 2025-02-15:
Yes, I deliberately made it like that, but that's only in gui. After long consoderation, I believe that a correctly working shifter has to be able to have at least this minimal range of axis, for it to function correctly.

Anyway, manualy you can set shifter calibration more precisely.

closed 2025-02-15 by NinjaFrozr