ranenbg/Arduino-FFB-wheel #86: Question about max gain and compilation

laurolopesdantas, 2025-02-11:
Hi Milos, first of all, great job man, really great updates, I tried your version 240x and was perfect to my project, and then I tried to make something odd, I jumpered de signal cables of the encoder on a g27 and wired the motors with the project in order to not tear down the g27 and keep it assembled, works fine, but due to the reduction gear on the encoder the ffb was too weak(way more weaker than the original g27), can you give me some advice were in the project I can set the max gain on the ffb above the maximum 200 that it has to compensate that, and how can I compilate the 0x version of the brWheel_my.ino (I already had the arduino IDE and etc, do not know how to select the version to compilate the 240x), Thanks in advance.

ranenbg, 2025-02-11:
Hi Lauro, thank you for copying the question in github. I get a lot of questions directly on my email and I can't keep repeating myself, so I hope you understand why I insisted to continue here.

Now about your question about max ffb gain. Obviously ffb signal has a limit and that is 100% (general gain slider). This is very much like audio signal level - where sound card can output only so much. You should not use more than 100% as this will lead to ffb clipping and loss of details. Instead try to increase ffb level from the source - game.

If you still need more, any further amplification is done by - amplifire, in our case this is the motor driver, or BTS7960. Since it operates in constant voltage mode, ffb signal amplitude is modulated using pwm (pulse width modulation) to drive the motor. Simply, a power supply voltage is quickly being turned on and off. The 0% pwm duty cycle corresponds to 0V, while 100% pwm is power supply voltage, say 12V. Motor coils have a certain resistance which for the sake of simplicity we're consider only as resistive (no complex impedance). By applying voltage across the coil some current will start passing and this current will create magnetic field in rotor which then interracts with a static mag field from magnets in stator and produce mechanical torque. Our ffb signal is this torque and it is in a direct relation with coil current.

If you want more ffb, you simply have to send more curent into the motor. This you can do by feeding more voltage to BTS, say 24V like original G29 motor driver circuit. Note that at 24V motor will draw more curent and power supply has to be able to provide that. This will also cause motor to dissipate some more heat, which you need to move away otherwise it will overheat and burn eventualy. I'm not sure, but G29 mainboard may have a constant curent motor driver and this is much better and it can directly control torque - ffb.

I have a section about compiling arduino source code in github readme towards bottom, have a look there if you're getting compiler errors. If you can compile, then go into config.h and simply comment out or uncomment certain lines. For v240x, you need only quad encoder and analog ffb axis, everything else is commented out. Correspondingly in the same header file a little bit down you need to set a version to 240, in hex format this is 0xF0. Correct version is required such that GUI can correctly recognize the specific firmware features and configure itself to properly display them.

laurolopesdantas, 2025-02-11:
Hi, I  understand your explanation but I made some tests and the weakness is related to de reduction gear on the g27 encoder, if I use a incremental encoder direct on the wheel shaft the ffb is fine, about the same of the original g27, but using the g27 optical encoder with the gear reduction the ffb end up divided by the ratio I guess. I will try to compile following your instructions, thanks for the answer

ranenbg, 2025-02-11:
You have to set the correct CPR including the gear reduction in GUI.You have to experiment a little bit untill you get 1:1 match between rotation of the virtual wheel in GUI and your G27 wheel rim.

I think encoder in G27 has 30 or 60PPR and gear reduction is 15:1. That gives a max CPR=4×60×15=3600, so try with this and adjust from there.

laurolopesdantas, 2025-02-11:
yes I did that in the beginning, 3600 and 4000 matches the angle but the ffb keeps weak, even with everything max out include in games, I think the code does not take into consideration the gear reduction, that same problem happens with other firmwares

ranenbg, 2025-02-11:
The firmware does take it into consideration if the CPR is set correctly. Even if it's not set correctly ffb does not depend on wheel angle, because the games send constant force ffb effect only. The ffb is weak probably because motors don't have enough curent.

laurolopesdantas, 2025-02-11:
12v 30A, as I said works fine if I place a incremental encoder directly on the shaft, only gets too weak using the geared optical encoder of the g27, how can I install the Arduino Boards v1.6.21 too compile the project ?

ranenbg, 2025-02-11:
Not sure why it makes a difference when mounting encoder on wheel shat as opposed to the motor via gear reductor.

In arduino IDE 1.8.19, go to manage boards then select 1.6.21 and install it.

Notriatv, 2025-02-12:
> 12v 30A, as I said works fine if I place a incremental encoder directly on the shaft, only gets too weak using the geared optical encoder of the g27, how can I install the Arduino Boards v1.6.21 too compile the project ?

g27 works with 24v motors

ranenbg, 2025-02-12:
What are you using to create ffb? Are you just using autocenter spring force or are you actualy testing in some game? If so, which game is it? You can also generate and test ffb with programs provided in misc folder like fedit or wheelcheck.

Enable realtime ffb monitor in GUI by pressing on a small red square button next to general gain slider. There you can see how strong the ffb signal is, coming directly from the game. If it's already in the red, increasing the gain will not do anything as ffb is already maxed out.

ranenbg, 2025-02-12:
That depends on how you connected the two motors. You have two options: [1] serial or [2] parallel connection.

For [1] the voltage will be split between motors, and for [2] current will be split between them. Neither of the options is ideal and the splitting will most probably be uneven. It will depend on the mechanical load that each motor sees, as well as coil temperature (resistance). You have to give both a try and observe the differences.

The ideal would be to use 2 bts and just provide the same PWM inputs to both of them.

laurolopesdantas, 2025-02-12:
The serial wiring is better, I  tried righ now and make a difference, the effects itself are not that much weaker, is more the resistance while turning that not maches the original g27 force, I tried to change the torque in PWD torque = torque*2; but seems to not make much difference.

ranenbg, 2025-02-12:
I see. Don't increase pwm signal by brute force, you will only get clipping.

ranenbg, 2025-02-12:
Not really, but you can ise pin11 for bts enable pins. I made such a feature to turn off bts when ffb is 0, so without autocenter spring it shouod be easier to turn.

laurolopesdantas, 2025-02-12:
But i want the opposite, the spring is already numb on turning, I want to boost up lol

laurolopesdantas, 2025-02-12:
I tried, make little to none difference, anyway thanks for the help, let me know in the future if you tried this odd behavior when the ffb end up been a lot weaker if using a low resolution encoder with a high gear reduction such that 1:15 one, I find this setup interesting because of the possibility of not tear down the original g27, and it is a lot more easy to build because the encoder is already in place and working, you can keep all your buttons, gearbox, pedals and etc, i think a lot of people would find this interesting.

ranenbg, 2025-02-12:
You are welcome.

Yes, that is the idea behind the project.

laurolopesdantas, 2025-02-12:
Don't ask me how, Out of the nowhere after some firmware reinstalling starts to work, now the force is the same as the original, not great but with a better and improved ffb, and I rewire the motors in parallel, thanks and again great job

laurolopesdantas, 2025-02-12:
for those who are interested in making this assembly is just jumper the first and the third pins on the encoder plug of g27 board(black and white wires) to the signals on the arduino, and then wire the motors to the bts(in parallel) and that is it! 

[image]

[image]

ranenbg, 2025-02-12:
You also need to connect ground wire from G27 board to the Arduino. How is encoder getting powered? Don't need to give it 5V to work?

laurolopesdantas, 2025-02-12:
don't need to connect the ground or the 5v, with the g27 on the usb it is already powered, as well the gearbox, pedals and all buttons, the advantage is exactly to use the wheel buttons and etc.

ranenbg, 2025-02-13:
Ah ok, you have a working G27 already :)

laurolopesdantas, 2025-02-19:
Yeap, today I notice a little problem with this build, de perfect CPR is 3945, I can turn the wheel to both ends and matches perfectly, I turn  all the way several time and the center keeps right, but if I turn the ffb on(BTS on psu) after a couple turns in game or in config it loses gradualy the center, there's no interference in the signal wires and only happen with the ffb on.

ranenbg, 2025-02-19:
Hi Lauro, yes, this is typical situation that we call "encoder step skipping".

Encoder needs pullup resistors, the lower the value the faster encoder response. Typicaly values between 10k and 1k do the trick. If you didn't put any pullups that explains the problem. I have enabled internal pullups in firmware but they are 100k which is simply not fast enough and additional ones are required.

ranenbg, 2025-02-19:
It has no pullups. That 1 resistor you see there is to limit the current on IR led. You need to put pullups between A - 5V and B - 5V.

laurolopesdantas, 2025-02-19:
Yeah I did that, equal what I do with the normal setup that I had, didn't work, 1k resistor

ranenbg, 2025-02-19:
Maybe it needs a pulldown resistor instead of pullup. The only way to know for sure is if you measure its output with a scope. Each encoder channel should have a nice square signal, where low level is 0V and high above 3V.

laurolopesdantas, 2025-02-20:
Tried pulldown and pullup and nothing in both cases the sensor turn off after do it, tune down all sliders to 0 except constant gain seems to help but the problem persist, very strange.

ranenbg, 2025-02-20:
No, ffb sliders have nothing to with it.

laurolopesdantas, 2025-02-20:
I was able to fix this issue by unplugging the encoder from de g27 board and providing a separated GND and 5v power to the cable, instead of take advantage of the g27 board 5v and gnd, thanks again, now this build is complete haha

ranenbg, 2025-02-20:
Great, I was about to suggest you the same thing.

closed 2025-02-20 by laurolopesdantas